samtap/fang-hacks #1: Wireless Connection Problem possible

jonnycastaway, 2017-01-14:
@samtap 
Hi, showed your hack collection here and found one litte problem.
You use the original iSC3S binary to establish a wireless connection. Thats no problem except you will use the Cam only local (like me) and block the path to the Internet because when iSC3S can't connect to the Manufacturer-Servers the wireless connection will also be aborted. So telnetd and so on is not accessible. Thats why i use the wpa-supplicant hack we described on [yi-hack](https://github.com/fritz-smh/yi-hack) all together.

samtap, 2017-01-14:
Yes it would be nice to take control of it, it's on my list to create a script. It is a bit difficult since when you screw up there's no way back into the device except serial console.. I don't think the average user will be able to solder wires to recover

samtap, 2017-01-26:
@jonnycastaway Can you try the new image and let me know if it solves your issue? When you don't use the cloud apps and you initially have no network connection at all, you'll not be able to apply the hack using the webinterface either, so you likely have to manually edit and rename config tmpl files in the bootstrap folder and edit fang_hacks.cfg.

jonnycastaway, 2017-01-30:
hi @samtap 
as you may have noticed i stayed away all these things fo a while. my 2 cams that i baked with no1knows and you runs very stable. and as you know i never used you repo because my idea was to run these cams without an sd-card. i used the glitchy way but all works fine on. so have fun with your stable base but i will do a step backward. when i buy a third cam i will test your superb work that brings all the little pieced together to have a great cam

samtap, 2017-01-30:
Closed, assuming resolved by V0.2.0

closed 2017-01-30 by samtap