samtap/fang-hacks #2: Stuck on 404

tobilap, 2017-01-14:
I got really exited once i saw your project after reading the discussion on the other github repo.
I'v setup everything the way you mentioned it in your readme.
100mb on /dev/sdd1  using mkfs.vfat
rest of the 16gb on /dev/sdd1 using mkfs.ext2

After inserting the SD card in the booted camera it does make a sound after a few seconds so I guess the runscript got executed (as I read in the yi-hack repo)

Sadly I'm getting a 404

The Camera is running on 2.8.3.5

samtap, 2017-01-14:
Can you use telnet to login? If so, please check the contents of /tmp/hacks.log

tobilap, 2017-01-14:
Sadly no, I did a portscan using nmap and telnet is closed. I just started preparing another sd card, maybe this will help.

samtap, 2017-01-15:
If you go to hello.cgi you should see a directory listing of the vfat partition, or opendir failed message is the sdcard isn't mounted

tobilap, 2017-01-15:
hmm, getting a 404 as well.

samtap, 2017-01-15:
Sure you're using the correct url? http://device/cgi-bin/hello.cgi

tobilap, 2017-01-15:
Arg. Was the SD card....works fine with the 3rd one i have set up...
Thanks for the help :)

closed 2017-01-15 by tobilap

dvbit, 2017-07-24:
might be interesting: 
rebooted
at first try I got connection refused
then 404