samtap/fang-hacks #5: Would it be possible to output rtmp stream (for Facebook/Youtube live video) from the device?

holleboom, 2017-01-16:
Hi, first of all: awesome repository, thanks a lot for the pre-baked image!

I'm trying to create a portable Youtube/Facebook live streaming device, would it technically be possible to run ffmpeg on the camera itself and output a rtmp stream [as described here](http://stackoverflow.com/questions/4010019/rtsp-to-rtmp-streaming)?

samtap, 2017-01-16:
I don't know, you'll just have to give it a shot yourself ;-).
My goal is to provide a stable base, for others to expand and collaborate on to add features together. Currently we're still depending on cloud apps, i.e. for getting the wifi up. So I'll be working on that for the time being.

You have to understand the CPU of the camera is very limited, all multimedia features are driven by hardware and don't use much CPU. Building ffmpeg is probably not worth it but perhaps you can create (or find) a crtmpserver build for arm architecture? Looks like at least it doesn't depend on ffmpeg nor has many other 3rd party dependencies so should be easy to build. No idea what's involved in converting rtsp -> rtmp and how CPU intensive it would be.

samtap, 2017-01-26:
Nginx RTMP module is included in new image. Can you test it? I've no ready made script yet so you have to experiment yourself.

holleboom, 2017-01-26:
Awesome, will try tonight!

sirmany, 2017-03-30:
Any advance of RTMP protocol with this cam?