samtap/fang-hacks #6: Add to synology surveilance station

mh2370, 2017-01-16:
Hi,

How to add the camera to synology surveilance station?
I have followed the guide here (for the Xiaomi ants camera)
But i can't get it working
https://forum.synology.com/enu/viewtopic.php?t=95211
Can you look at it?

Thanks

tobilap, 2017-01-16:
After writing the Image to the SD Card and setting up the scripts the RTSP server should run on 
rtsp://<DeviceIP>:554/unicast , if it doesnt work with your synology, try checking if it works for VLC.
If you just need other RTSP options just stop the rtsp server using the scripts(or killall) and run the server (data/usr/bin/manually/snx_rtsp_server) with different options. 
The server is documented here: https://github.com/haoweilo/RTSP_stream_server

roger-, 2017-01-17:
FYI if you have a working SDK and can build stuff, then this modified RTSP server adds authentication support (but is untested): https://gist.github.com/roger-/103f4c50ed63a660f26f63171b45b010

stratmetal, 2017-01-23:
This would be a great addition so we can monitor remotely with an authentication connection.

TheZoker, 2017-03-03:
@samtap Any news on that?

samtap, 2017-03-03:
I'm looking into this rtsp server which has builtin authentication but needs work to use with snx sdk: https://github.com/mpromonet/v4l2rtspserver 

I've been very busy with other stuff recently so no progress on the xiaofang camera. I rushed the image to hopefully attract others to develop stuff but so far no volunteers ;-)

gitingear, 2017-04-29:
I tested the code from roger (above) and authentication seems to work.
https://gist.github.com/roger-/103f4c50ed63a660f26f63171b45b010
in link above, only used lines 1-698, there is some extra code after that that should be omitted
also this is described in the code, but to clarify, the 20-rtsp-server script has to be modified to add -A option with _username_:_password_

roger-, 2017-04-29:
Awesome! I could never get the SDK to work properly so I never even got it to compile.

Could you share a binary?

gitingear, 2017-04-29:
replaced libsnx_rc.so in data/usr/lib, added snx_rtsp_server.auth2 in
data/usr/bin, chmod 755 on snx_rtsp_server.auth2, edit
data/etc/scripts/20-rtsp-server to point to new executable and add -A info
as needed

gitingear, 2017-04-29:
build env:
https://mega.nz/#!NWImyQBK!bMoytt2KtXJaGGCAjFdb-DQtM5IZS6REYSN22eXjcUI

desidude2000, 2017-05-01:
what's the password to the dev account in the above ova?

DavidRayner, 2017-05-01:
I don't know for this ova, but a different one that was shared on discord had the password devel.

desidude2000, 2017-05-01:
thanks DavidRayner.    i ended up booting to single-user and changing the password.   The ova is very useful.

DavidRayner, 2017-05-04:
There's an invite link at the bottom of the readme.

a-zimmerer, 2017-05-16:
Can  someone point me in the right direction how to get this started?
Is there a compiled version someone could provide me and give a
small tutorial which files to update. Thank you

a-zimmerer, 2017-05-18:
Okay, so I managed to compile rogers modified RTSP server on the sdk enviroment. It compiled flawlessly. Problem is if I replace the RTSP server on the sd card with the modified one and restart the service. Authentication works but i only get a blank screen. Any hints ?

desidude2000, 2017-05-19:
blockquote, div.yahoo_quoted { margin-left: 0 !important; border-left:1px #715FFA solid !important; padding-left:1ex !important; background-color:white !important; } Which device file under /dev are you pointing the binary to and trying to read the stream off of ?

a-zimmerer, 2017-05-19:
Not sure what you mean by "Which device file under /dev are you pointing the binary to" ? 
Like I said I just replaced the RTSP server on the SD card with the modified one. 

I am reading the stream with VLC rtsp://IP
It works perfectly with samtap's original version on SD card but not with the modified one.

a-zimmerer, 2017-05-19:
Do I only have to replace snx_rtsp_server or other files, too ?

rihotopis, 2017-05-20:
Hi

I understand that in order to compile the RTSP server I need to get the snx sdk from somewhere? Where could I get it?

ansco11, 2017-05-22:
@rihotopis 

https://mega.nz/#!514ERD7K!SEoFAnYDZ0-Ckolh4VAU64lTE2UR-as_9qw-1Q1-sWM

rihotopis, 2017-05-22:
Thanks a bunch!

rihotopis, 2017-05-22:
Hey @a-zimmerer 

Can you give any insight in how you set up the snx sdk for compiling the custom camera code?

ansco11, 2017-05-23:
For SDK 1.60 vanilla build of rtsp_server

cd /usr/src/sn986/snx_sdk/buildscript; \
        make sn98660_QR_Scan_402mhz_sf_defconfig
cd /usr/src/sn986/snx_sdk/middleware ; \
        mkdir -p _install/lib _install/include; \
        cp video/middleware/lib/lib* \
        rate_ctl/middleware/lib/lib* \
        sdrecord/middleware/lib/lib* \
        audio/middleware/lib/lib* \
        _install/lib; \
        cp -r video/middleware/include/snx_isp \
        video/middleware/include/snx_vc \
        rate_ctl/middleware/include/snx_rc \
        sdrecord/middleware/include/snx_record \
        audio/middleware/include/* \
        _install/include

PATH=/usr/src/sn986/snx_sdk/toolchain/crosstool-4.5.2/bin:${PATH}

cd /usr/src/sn986/snx_sdk/app/example/src/ipc_func/rtsp_server; \
        make

rihotopis, 2017-05-23:
@ansco11 

Thanks man!

a-zimmerer, 2017-05-23:
Problem is H.264 Stream ain't working with roger's code. Does anybody know how to get it running?

ansco11, 2017-05-23:
@a-zimmerer

What version of the SDK are you using?

a-zimmerer, 2017-05-24:
@ansco11 

Same Version everyone here uses. Are there more version in circulation ?
The source is compiling without error. Has to be something else.

Does the H.254 Stream work for you with rogers code. I can only get the MJPEG 
Stream working...

ansco11, 2017-05-24:
I've seen these versions:

1.60_QR_Scan_019a_20160606_0951
1.50_20160511_1053_d_054a_20160511_1320
1.50_037a_20151022_1049

1.60 compiles without error but segfaults with the stock libraries. I'll try it with the 1.60 libraries sometime in case that's the reason for it.

a-zimmerer, 2017-05-24:
I already tried that and it didn't work. The two libs I copied from the SDK to samtap's Image are:
libsnx_vc.so
libsnx_rc.so

Didn't change anything...

newsera, 2017-05-28:
Did anyone try this: https://github.com/mpromonet/v4l2rtspserver to secure the stream?

samtap, 2017-05-28:
@newsera Yes that's what I'm working on and hope to include in a next release

newsera, 2017-05-28:
That is an awesome news samtap 👍

1043717432, 2017-06-03:
@a-zimmerer  

Hi, have you managed to let Xiaofang store the video stream in SynologyNAS? Can you tell me how to do it? Thank you, and I need to do this to store Xiaofang video streams!

snflke, 2017-07-10:
@a-zimmerer which SDK did you use? I tried both 1.60_QR_Scan_019a_20160606_0951 and 1.50_20160511_1053_d_054a_20160511_1320. Both gave me segmentation fault. I tried static link, then got error about lgcc_s.so.

snflke, 2017-07-10:
Sorry for false alarm. snx_rtsp_server from SDK works fine. Original issue was because Filezilla corrupted the file during transfer. Just use Ubuntu command line ftp which transfer file without corruption.

ansco11, 2017-07-11:
@snflke Can you please confirm that any of these md5sum hashes are OK?

2314fffba7dcd777925b9bd45df468cd  SN986_1.50_037a_20151022_1049.tgz
3af7628fe2bc41ed2c4d21a4b1bf49c0  SN986_1.50_20160511_1053_d_054a_20160511_1320.tgz
e7a1de915e396419cf0e104577d190de  SN986_1.60_QR_Scan_019a_20160606_0951.tgz

snflke, 2017-07-11:
I have got 2 of these:
3af7628fe2bc41ed2c4d21a4b1bf49c0  SN986_1.50_20160511_1053_d_054a_20160511_1320.tgz
e7a1de915e396419cf0e104577d190de  SN986_1.60_QR_Scan_019a_20160606_0951.tgz

BTW, I merged in Roger's authentication code to SDK's rtsp_server. Clear text username/password worked fine. Now I can access my cameras using Tinycam, which reported "H264 SW".

DavidRayner, 2017-07-11:
Hmm like others here I tried using roger's code and was only able to get MJPEG to work. Can you share your snx_rtsp_server executable please?

snflke, 2017-07-11:
8935731ae541a8277bf1cc4d74045592  snx_rtsp_server
[snx_rtsp_server.zip](https://github.com/samtap/fang-hacks/files/1139855/snx_rtsp_server.zip)

start by modifying  /media/mmcblk0p2/data/etc/scripts/20-rtsp-server

start()
{
  LOG=/media/mmcblk0p2/data/mytest/rtsp.log
  echo "Starting RTSP server..."
  snx_rtsp_server -Q 5 -u media/stream1 -P 554 -A my_user:my_pass >$LOG 2>&1 &
  echo "$!" > "$PIDFILE"
}

snflke, 2017-07-11:
Tinycam setting:

Camera vendor: Generic
JPEG/MJPEG/RTSP request: rtsp://ip_address:554/media/stream1
Protocol: RTSP over TCP (MPEG/H264/H265)
Username: (in 20-rtsp-server)
Password: (in 20-rtsp-server)

DavidRayner, 2017-07-12:
Cheers, I've got authentication working using that. I get a segmentation fault if I specify a different width or height for the stream, although I want 720p which is the default so that doesn't matter.

The usage doesn't list a flag for audio, but when trying with -a I get the following error:

`snx_rtsp_server: symbol 'snd_lib_error': can't resolve symbol
snx_rtsp_server: symbol 'snd_pcm_extplug_create': can't resolve symbol
snx_rtsp_server: symbol 'snd_config_iterator_next': can't resolve symbol
snx_rtsp_server: symbol 'snd_pcm_extplug_set_slave_param_list': can't resolve symbol
snx_rtsp_server: symbol 'snd_config_iterator_entry': can't resolve symbol
snx_rtsp_server: symbol 'snd_config_get_id': can't resolve symbol
snx_rtsp_server: symbol 'snd_config_iterator_first': can't resolve symbol
snx_rtsp_server: symbol 'snd_pcm_extplug_set_param_list': can't resolve symbol
snx_rtsp_server: symbol 'snd_config_iterator_end': can't resolve symbol
ALSA lib dlmisc.c:252:(snd1_dlobj_cache_get) Cannot open shared library /usr/lib/alsa-lib/libasound_module_pcm_snx_audio_alaw.so
open cap snx_audio_alaw failed!!
Segmentation fault
`

Does audio work for you? Are you able to build the executable with the listed libraries linked? Thanks!

snflke, 2017-07-13:
Fixed. libasound should not be statically linked.
ad679a1d890981dc94427fc49d9b379c  snx_rtsp_server
[snx_rtsp_server.zip](https://github.com/samtap/fang-hacks/files/1143751/snx_rtsp_server.zip)

DavidRayner, 2017-07-14:
That works perfectly thanks! Just a note for others that use tinycam, you do have to use the 'Generic' camera vendor as it allows you write the full RTSP request. Tinycam can't load the stream when using the Xiaomi/Xiaofang set-up even if you configure it correctly.

For some reason tinycam doesn't allow you to specify a WAN IP to use when not connected to your home network (options are greyed out for a Generic cam), so I just entered my WAN IP as the main camera IP.

newsera, 2017-07-29:
@snflke @DavidRayner How to use snx_rtsp_server.zip to get authentication going? I assume the below file supports audio? 
ad679a1d890981dc94427fc49d9b379c snx_rtsp_server
snx_rtsp_server.zip

DavidRayner, 2017-07-29:
Extract the zip file to get the _snx_rtsp_server_ binary (which has no file extension). Copy this onto the camera replacing the usual binary located at _/media/mmcblk0p2/data/usr/bin/snx_rtsp_server_. Ensure new binary has 777 permissions once on the camera.

Then modify the text file on the camera located at _/media/mmcblk0p2/data/etc/scripts/20-rtsp-server_ so that it has a line similar to this
`snx_rtsp_server -Q 10 -u media/stream1 -P 554 -A my_user:my_pass -a >$LOG 2>&1 &`
instead of the usual line which executes snx_rtsp_server.

Jump on discord and ask for help if you need more detailed help.

newsera, 2017-08-03:
@DavidRayner Thank you so much.. It worked awesome..Do you know the default bitrate, width and height? Do you prefer RTSP over TCP or UDP?

louis-lau, 2017-08-03:
@newsera Width and height is locked to 1280x720 with this server. I think the default bitrate is 1024kbps.

I prefer TCP because I'm doing motion detection on the other end, and UDP wouldn't be good for that :P

newsera, 2017-08-03:
👍 Perfect :) Works good.. I think I will buy more cameras lol.. Have been waiting for authentication for a long time now..

ftruzzi, 2017-08-12:
My camera is much slower after using the snx_rtsp_server binary with authentication and sometimes reboots itself. Anyone experiencing the same issue? Sometimes it doesn't even start the RTSP stream.

comeollas, 2017-08-25:
Hi, and thanks to all, with this post first and fang hack i make work the xiaimi fang 1080 :)

When i see the rtsp is not protected i see this post and modified the rtsp-server.pid

But i make one error, i put -u media/stream1 copied from here but i see "not's a good idea" because i have all default directories and now after put user and pass doen't work rtsp strem.

anyone can help for make the correct default  directori or create working media/stream1?

comeollas, 2017-08-25:
Maybe use rtsp://myuser:mypass@"deviceip":554/unicast doesn't the correct form after put pass?
In VLC if i put only rtsp://deviceip/unicast work because i need to put pass and username but after put the correct pass and user say an error.

I tryed a lot of combinations and nothing :(

DavidRayner, 2017-08-28:
You don't need to worry about the media/* directories on the camera, the camera will handle them itself.

Inside my 20-rtsp-server file I have:

_snx_rtsp_server -Q 10 -u media/stream1 -P 554 -A my_user:my_pass -a >$LOG 2>&1 &_

For use with VLC I open the following stream:

_rtsp://my_user:my_pass@my_ip:554/media/stream1_

Are you able to access the camera from the same local network?

comeollas, 2017-08-29:
Yes, i log in with same local network (i use one laptop conected at same wifi without firewalls or rules, and port 554 works because without modify the snxserver all work without pass) but in the moment i set the snx_rtsp_server line i can't connect to rtsp with vlc.

I go to see another time the 20-rstp.... file for see if i used caps or not in -P.

comeollas, 2017-08-29:
Finally works, i erased -W1980 -H1080 (i don't remeber the exact but the image quality in this line i erased and works now!).

Tnx with your exact line of 20-rts... works fine.

gabryk91, 2017-09-13:
@DavidRayner thanks for the info, it works fine, but I have a problem with TinyCam. Before the mod I could connect and see the stream even on the WAN, now I can't. Do you know how to configure tinycam or similar program on an android device? I already tried vlc for android but it get stuck. Thank you

ardefy, 2017-09-14:
Hi, I it very good !!! thanks for this info !! :)

I also have problems with tinycam, it works well but the flow stops after a while, works in WAN and LAN with dyndns and ports redirected in NAT in my box. I think I would have to lower the quality of the rtsp stream. Is this possible with your server?

Regards :)

gabryk91, 2017-09-14:
Ardefy can you share some screenshots of your tinycam configuration?

ardefy, 2017-09-14:
[image]

:)

ardefy, 2017-09-14:
But the stream stop regularly :-(

DavidRayner, 2017-09-14:
do you have a stable and strong upload speed from the network the camera is on?

[10 more replies]