samtap/fang-hacks #7: After reboot

dmolner, 2017-01-17:
Hello,
Congratulations for the hack.
Is necessary extract the SD every time we reboot the cam?
Is possible execute the script after reboot with SD in the cam?

Best regards.

tobilap, 2017-01-17:
It is not required to extract the SD Card every time. Once set up you can leave it in the cam.

dmolner, 2017-01-17:
I think I have some worng.
I reboot the cam but I can't acces to the cgi webpage.
Is possible fix the IP of the cam?

tobilap, 2017-01-17:
The safest thing to do is to give your cam a static IP using your router.(Check the DHCP settings, you should be able to give a specific IP to a specific MAC address )

samtap, 2017-01-17:
@dmolner Can you use telnet to access /tmp/hacks.log?

dmolner, 2017-01-17:
@samtap this is the hacks.log

```
###/media/mmcblk0p2/data # vi /tmp/hacks.log
Mon Jan 16 07:33:18 UTC 2017 - /etc/fang_hacks.sh: Removed stale logfile
Mon Jan 16 07:33:18 UTC 2017 - /etc/fang_hacks.sh: Executing script (enabled: 1)
Mon Jan 16 07:33:19 UTC 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/action -> /tmp/www/cgi-bin/action
Mon Jan 16 07:33:19 UTC 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/action.orig -> /tmp/www/cgi-bin/action.orig
Mon Jan 16 07:33:19 UTC 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/func.cgi -> /tmp/www/cgi-bin/func.cgi
Mon Jan 16 07:33:19 UTC 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/scripts -> /tmp/www/cgi-bin/scripts
Mon Jan 16 07:33:19 UTC 2017 - /etc/fang_hacks.sh: Linking /media/mmcblk0p1/bootstrap/www/status -> /tmp/www/cgi-bin/status
Mon Jan 16 07:33:19 UTC 2017 - /etc/fang_hacks.sh: Running startup scripts
Mon Jan 16 07:33:31 UTC 2017 - /etc/fang_hacks.sh: Finished
```

dmolner, 2017-01-18:
@samtap 
I reboot camera with out SD and later I put SD.
Now I can acces to cgi web.
This is the hacks.log now.

```
### Mon Oct 10 16:00:24 UTC 2016 - /etc/fang_hacks.sh: Executing script (enabled: 1)
Mon Oct 10 16:00:24 UTC 2016 - /etc/fang_hacks.sh: Failed to find hacks in /media/mmcblk0p2/data!
mount: mounting /dev/mmcblk0p2 on /media/mmcblk0p2 failed: No such file or directory
Mon Oct 10 16:00:24 UTC 2016 - /etc/fang_hacks.sh: Failed to find /media/mmcblk0p2/data!
Wed Jan 18 08:15:08 UTC 2017 - /media/mmcblk0p1/snx_autorun.sh: Running (device: mmcblk0p1)
Wed Jan 18 08:15:08 UTC 2017 - /media/mmcblk0p1/snx_autorun.sh: Starting telnetd
```

gsbrown, 2017-01-18:
I just got the same error as dmolner.

I expanded the file system and then it said I had to reboot, so I did. Upon reboot, the status page would not load anymore, so I pulled power from the cam, removed SD card, waited for full boot, inserted SD card, heard the hammer noise and the status page now loads but no RTSP stream.

Here is my hacks.log:
`Mon Oct 10 12:00:24 GMT 2016 - /etc/fang_hacks.sh: Executing script (enabled: 1)
Mon Oct 10 12:00:24 GMT 2016 - /etc/fang_hacks.sh: Failed to find hacks in /media/mmcblk0p2/data!
mount: mounting /dev/mmcblk0p2 on /media/mmcblk0p2 failed: No such file or directory
Mon Oct 10 12:00:24 GMT 2016 - /etc/fang_hacks.sh: Failed to find /media/mmcblk0p2/data!
Wed Jan 18 10:29:07 GMT 2017 - /media/mmcblk0p1/snx_autorun.sh: Running (device: mmcblk0p1)
Wed Jan 18 10:29:07 GMT 2017 - /media/mmcblk0p1/snx_autorun.sh: Starting telnetd
`

tobilap, 2017-01-18:
Did you check if the filesystem expansion worked correctly?
After pressing the expand button do not reboot immediately, it takes some time to expand. If you reboot to early you are breaking the ext2 partition and you'll get the errors.

You can also expand the filesystem manually e.g. with gparted.

gsbrown, 2017-01-18:
hmm, I waited some time but did not wait for any confirmation. That was probably what I did wrong. What is the best way to fix it? (noobish) Do I just re image the sd card and start over?

tobilap, 2017-01-18:
Easiest way is to start over and wait 10 minutes after pressing the expansion button. Or if you are running linux and more into GUIs you can easily use gparted to do the expansion manually.

samtap, 2017-01-18:
The expansion is done in two steps:
1. Update partition table, occurs when you click the status page and is finished when you get the output.
2. Expand the filesystem, occurs during reboot and takes quite a while depending on sd-card size. When you open status page you'll get error about hacks not found in /media/mmcblk0p2. Resize2fs will be shown in process list. Wait a while, refresh until /media/mmcblk0p2 is mounted and the error on status page disappears.

There're some issues with boa mainly caused by the non-deterministic startup. This will be improved when I disable cloud apps from starting and manually start boa myself, in the next release. Boa also sometimes fails to start (unable to access boa.conf or some error like that), if you copy all files from /usr/boa to a different location (/tmp) it works fine. Also I suspect /tmp/www is removed and a fresh copy of hello.cgi is put in place as part of the boa startup procedure.

Meanwhile, if you often find cgi scripts are not available after a clean boot, you could add a startup script to manually start boa and copy the section to create symlinks to cgi-scripts from fang-hacks.sh

gsbrown, 2017-01-18:
Awesome work on this, I just received my cameras yesterday.  I will give it another go; next up will be getting the stream  to display properly in home assistant. So far only snapshots are displayed. (Home assistant uses ffmpeg to display the rtsp stream).

gsbrown, 2017-01-18:
I am not sure how to start over properly. I re-imaged the sd card and popped it in the camera, heard the noise and it showed the option to extend the filesystem, I selected apply and right away it told me the filesystem was already extended and that I have to reboot. Now I am getting the same results. Is there some file I have to remove from the camera as well?

samtap, 2017-01-18:
Remove /etc/fang_hacks.sh if you want to re-apply the hack. But it seems you're only having issues with sdcard partitions. If you re-imaged the sdcard and the expand filesystem reports the partition is already at max size, something went wrong with flashing the image to sd-card. 

After flashing the image, here's a hidden file '.resize' on the data partition. If this file exists, the status page will show the option to expand it. If the partition table is succesfully updated, a hidden file '/etc/.resize_runonce'  triggers the filesystem expansion (resize2fs) upon reboot (and status page will show reboot is required). 

That's all, very simple really. Just look into the scripts they should explain themselves :)

dmolner, 2017-01-21:
@samtap 
I removed the fang_hacks.sh and write new the SD.
The first time all work well I can connect by telnet and acces to web. I expand partition and wait finish and 10 minuts more. But after reboot not work web page and only work rtsp server.

noname1020, 2017-01-24:
Same here,after reboot it will not start the services,to solve this issue I simply edit /etc/init.d/rc.local sleep 30 to sleep 65 and it work every time when reboot or do power cycling.

dmolner, 2017-01-25:
@noname1020 Great.
Works perfect.

samtap, 2017-01-26:
In the latest image all excessive sleeps have been removed and the scripts use polling loops to wait for cloud apps. Feel free to re-open if you're still having timing related problems.

closed 2017-01-26 by samtap