samtap/fang-hacks #8: Failing to apply scripts - missing files?

frode, 2017-01-17:
Applying fang hacks scripts...
Source file not found: '/media/mmcblk0p1/bootstrap/sdcard'
Source file not found: '/media/mmcblk0p1/bootstrap/rc.local'
Source file not found: '/media/mmcblk0p1/bootstrap/fang_hacks.sh'

Device is running firmware v2.8.0.0. Latest is 2.8.3.5 according to the Mi Home app.

tobilap, 2017-01-18:
Sounds like you do not have the bootstrap folder placed on the first partition(FAT).
Did you use the Image from the release or did you prepare the partitions yourself?

ngtwolf, 2017-01-25:
So, i'm having the same problem here.  I used win32image to create the image onto an SDCard.  After telneting into it, and poking around, i see that the sdcard appears to be mounting as 'mmcblk1p1', not 'mmcblk0p1'.  Not sure why it's a different mount point then others, or what the easiest way to resolve this?

ngtwolf, 2017-01-26:
Okay, figured it out..  every time you put the SD Card in, it assigns the sd card as the next mount point.. I must have put the sd card in a 2nd time, so the mount point changed... i figured this out because i went and changed the file, then stuck it in the camera again, and it changed it to the next letter... and so on... so yeah, it only works the first time you put it in the camera after powering it on.  So, I'm good now, but maybe there might be some error checking at some point :)

samtap, 2017-01-26:
I've mentioned this in the updated README. It now instructs users to boot without sd-card and insert it later. It only happens when the sdcard is in use when you remove it.

closed 2017-01-26 by samtap

ngtwolf, 2017-01-28:
Thanks.  Yup, I was using the older version and noticed the newer version when I was messing with things so probably didn't catch the updated readme.  I'm still having a few random quirks i'm trying to work out when i get a chance, but in any case, thanks so much for all the work you and everyone have done...  If I can get this all set up with the Zoneminder setup i've been working on (on RPi, which may have not been such a good idea..haha), I can get a few more of these cameras and ditch the lousy dlink ones I have.