samtap/fang-hacks #9: Can't apply hacks

fallenruby, 2017-01-18:
Hi, I think I have made some mistake in the process.

from the status page I have this warning:
`Script is already installed!
Error: No 'data' directory found in /media/mmcblk0p2! `

here's my logs:

`Contents of /tmp/hacks.log:
Wed Jan 18 17:04:36 GMT 2017 - /etc/fang_hacks.sh: Removed stale logfile
Wed Jan 18 17:04:36 GMT 2017 - /etc/fang_hacks.sh: Executing script (enabled: 1)
Wed Jan 18 17:04:36 GMT 2017 - /etc/fang_hacks.sh: Not linking /media/mmcblk0p1/bootstrap/www/action: already exists
Wed Jan 18 17:04:36 GMT 2017 - /etc/fang_hacks.sh: Not linking /media/mmcblk0p1/bootstrap/www/action.orig: already exists
Wed Jan 18 17:04:36 GMT 2017 - /etc/fang_hacks.sh: Not linking /media/mmcblk0p1/bootstrap/www/func.cgi: already exists
Wed Jan 18 17:04:36 GMT 2017 - /etc/fang_hacks.sh: Not linking /media/mmcblk0p1/bootstrap/www/scripts: already exists
Wed Jan 18 17:04:36 GMT 2017 - /etc/fang_hacks.sh: Not linking /media/mmcblk0p1/bootstrap/www/status: already exists
Wed Jan 18 17:04:36 GMT 2017 - /etc/fang_hacks.sh: Failed to find hacks in /media/mmcblk0p2/data!
mount: mounting /dev/mmcblk0p2 on /media/mmcblk0p2 failed: Device or resource busy
Wed Jan 18 17:04:36 GMT 2017 - /etc/fang_hacks.sh: Failed to find /media/mmcblk0p2/data!`

I can't stream nor FTP into the camera.

samtap, 2017-01-18:
According to the log, there's no data folder on your mmcblk0p2 ext2 partition. Did you create the sdcard by flashing the image?

fallenruby, 2017-01-18:
Yes, downloaded and burned with Win32disk imager. I have tried to write the image several time and different SD card.

samtap, 2017-01-18:
Only thing I can think of is that it got mounted as mmcblk1 instead of mmcblk0 when you re-insert it to apply the hack. But that should be resolved after a reboot.

fallenruby, 2017-01-18:
Mounts:

rootfs on / type rootfs (rw)
/dev/root on / type cramfs (ro,relatime)
none on /proc type proc (rw,relatime)
none on /sys type sysfs (rw,relatime)
none on /proc/bus/usb type usbfs (rw,relatime)
dev on /dev type tmpfs (rw,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,relatime,mode=600)
/dev/mtdblock4 on /etc type jffs2 (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime,size=36956k)
lock on /var/lock type tmpfs (rw,relatime)
log on /var/log type tmpfs (rw,relatime)
run on /var/run type tmpfs (rw,relatime)
spool on /var/spool type tmpfs (rw,relatime)
tmp on /var/tmp type tmpfs (rw,relatime)
mq on /var/mq type tmpfs (rw,relatime)
media on /media type tmpfs (rw,relatime)
/dev/mmcblk0p1 on /media/mmcblk0p1 type vfat (rw,relatime,fmask=0000,dmask=0000,allow_utime=0022,codepage=cp437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p2 on /media/mmcblk0p2 type ext2 (rw,relatime,errors=continue,user_xattr,acl)

samtap, 2017-01-19:
Looks fine, no idea how the data folder got removed.

fallenruby, 2017-01-19:
OK,  can you give me direction for starting over clean?

closed 2017-01-20 by samtap