samtap/fang-hacks #10: Firmware v3 changes

gonzonick, 2017-01-19:
I have the v3 firmware running on my camera and it has a few changes from v2. For reference, I started with the v0.0.1 sd card release. Here are the things I noticed or had to tweak to get it running on my v3 firmware vs the v2 firmware:

- There is no iSC3S process, it appears to now be iCamera (two instances of this is running, and lsof shows them using /dev/video0 and /dev/video1)
- You have to kill -9 (not just kill) to get both of the iCamera processes to stop. Once you stop them, the camera will restart within 30 seconds. 
- The restart seems to be tied to the watchdog module, which does not unload with rmmod (or else the 00-stop-cloud startup script would have taken care of it). I had to comment out where it gets loaded in /etc/init.d/rcS
- There are two scripts to start up rtsp, 02- and 03- in the data folder. I have removed the 02 script as it seemed redundant.

Once I stopped the snx_wdt module from loading, then modified the 00- script to kill iCamera and used 03- to start up rtsp I have a stable stream coming from the camera running 3.0.3.53 firmware. 

I understand and like the idea of making as few changes to the core system as possible, but I wasn't aware of any other way to stop the snx_wdt module from running without commenting it out on the actual device itself vs running a script post-boot. Maybe someone can figure a better way to take care of that.

samtap, 2017-01-19:
Haven't had any time but will try V3 myself during the weekend hopefully. Thanks for the info! Did you try modprobe -rf to get rid of snx_wdt?

gonzonick, 2017-01-20:
Just tried, yup modprobe -rf snx_wdt works to unload the module. I've updated the 00 script on my card to do that instead. Thanks.

The rtsp server crashes every few hours for me, I wrote a little do:while script to check it and launch it again if it isn't running but I am not sure if this instability is new to v3 or just the way it is. For now it works well enough just getting restarted, but it might not work once I try to get a constant stream and record it with a DVR.

navigonic, 2017-01-21:
Has anyone noticed some of the cameras firmware updating to 3.0.3.53 while others remaining at 2.9.0.7?

phtp, 2017-01-22:
I am running 2.8.3.5 and Mi Home prompted me couple days ago to update, but I ignored it. Now it's not longer prompting me and I noticed the Mi Home app updated instead.

samtap, 2017-01-26:
Latest updates add V3 support, closing this issue.

closed 2017-01-26 by samtap