samtap/fang-hacks #11: Source and documentation folders

petero-dk, 2017-01-19:
Hi @samtap 

I have forked your repository https://github.com/imaginarydenmark/fang-hacks/commit/4a03b9750018f1b7cd6a03af818ef0f72067f621 

And tried to move the files into a subdirectory called dist (did not go too well). 

I would like to add whatever developer documentation about the board and device additionally I would like to add the source code for tools compiled for the board as well.

I have been gaining access to more information about the board and I will be getting more. I need to figure out how to share as much as I can.

samtap, 2017-01-20:
@petero-dk of course you can do whatever you want, that's the nature of github. Please use this area only to report issues

closed 2017-01-20 by samtap

petero-dk, 2017-01-20:
@samtap yes i get that.

In the longer run I would obviously create a PR so there was just one "official" if you were interested in that. However because of the nature of the files (too simple?), they are not simple git mv, they will show up as file removed and file added. Which is why I would like to hear your take on it in order not to mess things up going forward.

Regards

Peter

samtap, 2017-01-20:
I have no desire to host any SDK documentation on my github, sorry.