samtap/fang-hacks #13: iSmartalarm Spot

mh2370, 2017-01-20:
Hi,
Is it possible to make work this Xiaofang camera with the iSmartalarm app
Because it is the same camera as the iSmartalarm Spot

https://www.ismartalarm.com/devices/cameras/spot/isa00013.html

Maybe with an edited firmware?

Thanks

Gilgameshismist, 2017-01-20:
Is not possible without the original iSmartalarm firmware. (which currently isn't available).

mh2370, 2017-01-20:
Ok i didn't know that
Hopefully can it be installed in the future
The iSmartalarm app is really nice

closed 2017-01-23 by samtap

snailium, 2018-03-16:
I'm able to find iSmartAlarm firmware v2.4.9.6, which seems to be the latest. It is from Android app.

I uploaded the firmware here.
https://1drv.ms/u/s!AjUVjow4El8Wn7MHpB6e5KZm1H-hYg

@samtap Please take a look if you have time. :)

Swiftnesses, 2018-11-21:
Amazing find, can we get this firmware uploaded to the camera and use it with the app?