samtap/fang-hacks #14: Have two different night scripts for LED and BW modes

Kryten2X4B523P, 2017-01-22:
Can we split the script for IR control into two scripts, one to control the LEDs when it is dark and the other switching the image to BW? This would be useful for a camera pointing out a window onto a street where the street lighting is enough for a clear BW image but not enough for a clear colour image, but obviously the LEDs reflect off the glass.

Looking at what has been posted in the hacks already, it may be easier to do this as one script for standard night mode with both LEDs on and BW image and another for this scenario turning image to BW but keeping LEDs turned off, and then just enable whichever script is more appropriate for individual needs.

magnets110, 2017-01-23:
I've been testing this (IR cut filter off + LEDs off) so that you can place it near a reflective surface.

To get the IR cut filter to move off you need to trigger the LEDs first. Then you can disable the LEDs and leave the filter off.

E.g. 
`gpio_ms1 -n 2 -m 1 -v 0
gpio_aud write 1 0 1
gpio_aud write 1 0 0`

then obviously change saturation

To go back to day mode
`gpio_ms1 -n 2 -m 1 -v 1`

Night mode without LEDs is fairly awful unless you have some other strong lighting, maybe there's an option to increase shutter speed?

Kryten2X4B523P, 2017-01-23:
Night mode without LEDs is useful if you are pointing out a window to a lit street. I have tested this on another camera by blocking the LEDs as well as reducing another camera to 5fps to increase exposure which although improved the night image considerably, still isn't as good as in BW night mode.

Obviously if we can do a longer exposure as well then that would be another option to offer as depending on other light sources (eg. if the street light is right outside your house) you might get a good clear full colour image that way.

Kryten2X4B523P, 2017-01-23:
Do you really have to trigger the LEDs first though? Doesn't [this comment](https://github.com/fritz-smh/yi-hack/issues/118#issuecomment-271917657) show that you can check if it is day or night without turning the LEDs on?

So wouldn't it simply be removing the LED switch from the script in [this comment](https://github.com/fritz-smh/yi-hack/issues/118#issuecomment-272433769) to do what I have proposed without ever turning on the LEDs, as below?

> #!/bin/sh
> echo "IR script started"
> # ir_init
> gpio_ms1 -n 2 -m 1 -v 1
> gpio_aud write 1 1 0
> gpio_aud write 0 2 1
> gpio_aud write 1 0 0
> 
> sleep 3
> 
> # ir loop
> IR_ON=0
> 
> while :
> do
>         DAY="$(gpio_aud read 2)"
>         if [ $DAY -eq 1 ]
>         then
>                 if [ $IR_ON -eq 1 ]
>                 then
>                         gpio_ms1 -n 2 -m 1 -v 1
>                         echo 0x40 > /proc/isp/filter/saturation
>                         IR_ON=0
>                 fi
>         else
>                 if [ $IR_ON -eq 0 ]
>                 then
>                        echo 0x0 > /proc/isp/filter/saturation
>                         gpio_ms1 -n 2 -m 1 -v 0
>                         IR_ON=1
>                 fi
>         fi
>         sleep 3
> done

Kryten2X4B523P, 2017-01-23:
Tested my script above just now, it works but still not as clear as just covering the LEDs on my Digoo camera, it would appear the Digoo also increases exposure in night mode. Swapping to BW only without LEDs did make the image a little clearer if only because it removed the bad colouration of a normal exposure in low light on this camera, but it certainly needs a longer exposure to really see a big improvement.

One other thing I have found useful, perhaps specific to me pointing a camera out the window, is having a 'force day mode' script. Basically it forces all settings to normal for day use, handy if like me you are testing the scripts at night and one turns on the LEDs. Running the 'force day mode' script once after deactivating your test script resets all behaviour.

jonnycastaway, 2017-01-30:
I tested around with this stuff a while ago when i searched an found the options for led, ircut and saturation.
And i never found a great way to get night pictures totally cool. the best way i found (when you see to a enlightened street etc.) is to let the camera in daymode.

Barnabasek, 2018-02-01:
You can control other parameters of the camera picture. After tweaking those you might be able to get a decent night picture.

echo 0x40 > /proc/isp/filter/saturation   #saturation
echo 0x40 > /proc/isp/filter/brightness   #brightness
echo 0x60 > /proc/isp/filter/gamma   #gamma correction
echo 0x0f > /proc/isp/filter/hue   #hue
echo 0x0f > /proc/isp/filter/contrast   #contrast
echo 0x1f > /proc/isp/filter/sharpness   #sharpness

(I don't know what are the default values. These just work for me at the moment).