samtap/fang-hacks #15: Time-lapse possibility

mstraa, 2017-01-23:
Hey, first thanks very much for the hacks! 

I  bought 2 xiaofang for time lapse purpose. How do you think I can do that?

tobilap, 2017-01-24:
The easiest way for you would be to setup a small server(e.g. raspberry pi) and use ffmpeg on it using the RTSP streams from the 2 cameras. Your could just start the rtsp server with -F 1 which reduces the FPS to 1 as you want to do a timelapse anyway.
The Camera itself does not have much CPU power, most of the things have their own hardware units( e.g. the video encoding) to be efficient. It does not require lots of CPU power.

tobilap, 2017-01-24:
You can use any setting you want, Increasing the resolution up to 1920 * 1080 but important is that you increase the bitrate (e.g. -b 2048 on 720p streams) to get good quality Images delivered.

samtap, 2017-01-24:
I think you will see that when connecting to snx_rtsp_server, the first frames are always garbage... Very inconvenient for taking a snapshot.

tobilap, 2017-01-24:
How about MJPEG instead of h264 as it compresses every single frame?

magnets110, 2017-01-24:
You can try decreasing the GOP size to stop the junk frames at the start of the stream. -g 10 on the snx rtsp server

I believe this will result in worse compression rates 

The camera should be powerful enough to dump the stream to disk, there isn't any encoding required. The camera does motion detection in hardware so you can pull from that. I think samtap is working on doing something like that

mstraa, 2017-01-25:
So I tried with those params : snx_rtsp_server -W 1920 -H 1080 -j 25 -Q 1 -F 1
I get a good quality of my desk (>1m) but when I turn the cam to the room, quality decrease and I get some frames (not all) with an error with a buffer size : 

 `MultiFramedRTPSink::afterGettingFrame1(): The input frame data was too large for our buffer size (263512).  71124 bytes of trailing data was dropped!  Correct this by increasing "OutPacketBuffer::maxSize" to at least 333268, *before* creating this 'RTPSink'.  (Current value is 262144.)`

mstraa, 2017-01-25:
Well the rtsp stream works fine with vlc but I can't get it with ffmpeg.

Server :
`snx_rtsp_server -W 1920 -H 1080 -j 25 -Q 1 -F 1`

Ffmpeg :
`ffmpeg.exe -rtsp_transport tcp -i rtsp://192.168.1.164/unicast`

Error : 
`Trailing options were found on the commandline.
[rtsp @ 000000000030a400] Invalid RTP/JPEG packet. Quantization tables not found.
[rtsp @ 000000000030a400] Received packet without a start chunk; dropping frame.
    Last message repeated 118 times
[rtsp @ 000000000030a400] Invalid RTP/JPEG packet. Quantization tables not found.
[rtsp @ 000000000030a400] Received packet without a start chunk; dropping frame.
    Last message repeated 126 times
[rtsp @ 000000000030a400] Invalid RTP/JPEG packet. Quantization tables not found.
[rtsp @ 000000000030a400] Received packet without a start chunk; dropping frame.
    Last message repeated 127 times
[rtsp @ 000000000030a400] Invalid RTP/JPEG packet. Quantization tables not found.
[rtsp @ 000000000030a400] Received packet without a start chunk; dropping frame.
    Last message repeated 127 times
[rtsp @ 000000000030a400] Invalid RTP/JPEG packet. Quantization tables not found.
[rtsp @ 000000000030a400] Received packet without a start chunk; dropping frame.
    Last message repeated 123 times
.`

mstraa, 2017-01-28:
I tried few ffmpeg bin (32bit 64bit / 3.2.2 Nightly) and i get always the same error. 
Can you tell me wha4t params do you run the rstp server ?

virualdo, 2017-08-11:
@mstraa My ffmpeg script was working fine until i've updated the snx_rtsp_server file with the version that have rtsp and jpg streams.
https://github.com/samtap/fang-hacks/issues/87#issuecomment-318074811
When i updated that file i've started to have this error. I've tried with other ffmpeg files in my pc without any luck. I think that the rtsp stream have some kind of incomplete metadata and ffmpeg process fails when try to use it.