samtap/fang-hacks #18: Custom WiFi settings not applied when cloud apps are running

samtap, 2017-01-26:
There's an issue in v0.2.0 that causes the custom WiFI modes (WiFi Client / AP mode) to not get applied when cloud apps are used (DISABLE_CLOUD=0 and 00-stop-cloud script disabled).

The cloud apps take quite a while to get the interface up and the custom script that sets up the interface runs too early. It sets up the interface correctly, but the settings are lost as soon as the cloud apps re-configure it, using credentials from /etc/config instead of custom configuration from /media/mmcblk0p2/data/etc.

As a short term workaround you can insert a sleep in the 01-network script, to make it wait until the cloud apps are done before overriding the interface with custom configuration. A proper fix will be in the next release.

camtheram, 2017-01-27:
Hi - where can I share my ideas / feature requests? (I'm not a coder, I can't develop them.) Thanks for your work!

samtap, 2017-01-27:
@camtheram Feel free to open issues for feature request. Don't be mad if I close it because I think it's not useful or too difficult or whatever. I've just enabled wiki edit access so you should also be able to write and discuss ideas there.