samtap/fang-hacks #19: OSD date stamp

magnets110, 2017-01-27:
ref: https://github.com/fritz-smh/yi-hack/issues/118#issuecomment-273896219
snx_isp_ctl

It allows an onscreen timestamp which is useful if you want to use it as an IP camera
This is the settings I am using:
`snx_isp_ctl --osdset-en 1 --osdset-ts 1 --osdset-template 1234567890./-:Date --osdset-gain 2 --osdset-bgtransp 0x1 --osdset-bgcolor 0x000000`

`gain 2` doubles the size of the font

The first line seems reserved for a string (--osdset-datastr), so it's always 1 line offset. There's other options for mirror, privacy mask etc.

`Usage: ./snx_isp_ctl [options]
Supported options:
        -c, --channel                           0: ISP0(default), 1:ISP1
         --osdset-en                    Set OSD Enable(1: Enable, 0: Disable)
         --osdget-datastr               Get OSD Data String
         --osdset-datastr               Set OSD Data String
         --osdget-ts                    Get OSD TimeStamp en
         --osdset-ts                    Set OSD TimeStamp en
         --osdget-gain                  Get OSD Gain(0~7)
         --osdset-gain                  Set OSD Gain(0~7)
         --osdset-txtcolor(0xFFFFFF)    Set OSD TXT Color
         --osdset-bgcolor(0x000000)     Set OSD Backgroud Color
         --osdset-txttransp             (yes: 0x0, no:0x1)      Set OSD TXT Transparency
         --osdset-bgtransp              (yes: 0x0, no:0x1)      Set OSD Backgroud Transprency
         --osdget-position              Get OSD Position
         --osdset-position      x,y     Set OSD Position
         --osdset-template              Set OSD Template
         --osdset-font                  Set OSD Font
         --osdget-width                 Get OSD Width
         --osdget-height                Get OSD Height

         --mdget-en                     Get Motion Detection Enable
         --mdset-en                     Set Motion Detection Enable(1: Enable, 0: Disable)
         --mdget-thre                   Get MD threshold(0~65535)
         --mdset-thre                   Set MD threshold(0~65535)
         --mdget-int                    Get MD Int
         --mdget-inttimeout             Get MD Int Timeout
         --mdset-inttimeout             Set MD Int Timeout
         --mdget-intthre                Get MD Int Threshold
         --mdset-intthre                Set MD Int Threshold
         --mdget-blkmask                Get MD Block Mask
         --mdset-blkmask                Set MD Block Mask
         --mdget-blkrepo                Get MD Block Mask

         --mfset-mode                   Set Mirror Flip Mode(mode:off:0, flip:1, mirror:2, flip&mirror:3)
         --aecset-en                    Enable ISP AEC(auto exposure control)enable(enable:1,disable:0)
         --awbset-en                    Enable ISP AWB(auto white balance)(enable:1,disable:0)

         --sensorset-mir                Enable sensor mirror(mode:enable 1, disable 0)
         --sensorset-flip               Enable sensor flip(mode:enable 1, disable 0)
         --sensorset-aec                Enable sensor AEC (mode:enable 1, disable 0)
         --sensorset-exp                Set sensor exposure time(exp:0x1 ~0xffff)
         --sensorset-awb                Enable sensor AWB(val:enable 1, disable 0)note:when use ISP AWB, should disable sensor AWB
         --sensorset-rgain              Set sensor red gain(gain:0x0000~0xffff)
         --sensorset-ggain              Set sensor green gain(gain:0x0000~0xffff)
         --sensorset-bgain              Set sensor blue gain(gain:0x0000~0xffff)
         --sensorget-name               Get current sensor name

         --hdrset-en                    Enable hdr (enable 1, disable 0)
         --drcget-status                Get drc status(enable 1, disable 0)
         --drcset-status                Set drc status(enable 1, disable 0)
         --drcget-value                 Get drc intensity value
         --drcset-value                 Set drc intensity value(val:0~15)

         --pmset-en                     Enable private mask(enable:1,disable:0)
         --pmset-color                  Set private mask block color(color:rgb888 format)
         --pmset-area                   Set private mask area(area:[6], 6*32bit represent 192blocks)(0x11 0x11 0x11 0x11 0x11 0x11)

         --filterset-contrast           Set image contrast (val range is [0~64])
         --filterset-sharp              Set image sharpness (val range is [0~6])
         --filterset-sat                Set image saturation (val range is [0~127])
         --filterset-hue                Set image hue (val range is [0~359])
         --filterset-bright             Set image brightness (val range is [0~127])
         --afget-sum                get auto focus w0-sum (200ms + period time)
`

samtap, 2017-01-27:
Thanks, moved this to the wiki and enabled edit access for anyone. Feel free to add more content!

closed 2017-01-27 by samtap

Stalker13, 2017-03-09:
Hi. What is parameter --osdset-font? I think it's used to set the type of font, but what values can be set in this parameter? Thanks