samtap/fang-hacks #20: Portforwarding with Unicast

PipoPT, 2017-01-28:
Hi there.
Been messing around and with the help from people in the chat I could make this work on vlc.
Since, I'm trying to forward the ip address of the camera to a public domain (so I can check the camera from any network), but I think I'm missing something.

I can't find a way to forward the RTSP address. Can anyone give a hint?
Thanks in advance

ngtwolf, 2017-01-28:
So, there are a lot of pieces to this.  I'm not sure which part you're getting stuck on.  

First, you'll need to forward the domain to your home IP address.  If your IP changes (like any consumer internet service), you'll be better off setting up an account on a Dynamic DNS site, then forward your domain to that dynamic dns hostname.  My recommendation would be 'freedns.afraid.org'.  This involves you installing software however, so the easiest thing is to look at your router under the dynamic dns section and see what it supports and go with that.

Second, you'll need to set up port forwarding.  There should be a section on your router where you set an internal (554) and external port (whatever you want), as well as internal IP of your camera.  You'll likely not want to use the internal 554 RTSP port as the external port, but you can.  

Third, in VLC, put your external IP (you can get that by going to 'whatismyip.com') and the forwarded external port you used (rtsp://<MYEXTERNALIP>:<MYEXTERNALPORT>/unicast).. this should allow you to at least test it. You'll then want to do test with all the dynamic dns info mentioned above.

Caveats..  All this above is a VERY BAD idea.  This will be putting your camera on the internet and completely unsecured to anyone that stumbles across it.  The reason most people are doing this is because they don't trust the chinese xiaomi servers having access to their cameras, but this method is  giving access to the entire world... I would use the Xiaomi app before I ever did this.  The other problem is that if you want to do this more than one camera, you have to use multiple ports each one pointing to the IP and 554 port on the internal camera and it will become a complicated mess.  My suggestion would be to use a camera server software (Zoneminder is open source) that has authentication built into it, and point to that instead of your cameras directly.

jonnycastaway, 2017-01-30:
I had similar problems with my VPN connection because the UDP-Packets of rtsp (which contains the video) can't go trough the vpn. so i had to use the -t parameter for rtsp over tcp:

If anyone is willing to stream in tcp (because of vpn behaviors like me) then switch the line in rc.local
/etc/rtsp/snx_rtsp_server -W 1920 -H 1080 -Q 15 -b 4096 -a -P 554 -T 8554 &

-P is the UDP Port (0 = disabled)
-T is the TCP Port (0 = disabled)

eventually @samtap can insert a switch in his control site?

samtap, 2017-01-30:
There's always going to be somebody who's not happy with the default I choose. I'm not going to make every single thing configurable either (I'm not a webdeveloper). You'll just have to edit the script to your personal preferences.

closed 2017-02-02 by samtap