samtap/fang-hacks #21: FH V0.2.0 path to hack broken

mstraa, 2017-01-28:
Hi !

I have to Xiaofang and one is on the 3.0.3.56fw. When I put sdcard with FH 0.2.0 I can go to the cgi page but I cannot apply patch, for some reason it mount on mmcblk1pX:

Error 
Applying fang hacks scripts...
Source file not found: '/media/mmcblk0p1/bootstrap/sdcard'
Source file not found: '/media/mmcblk0p1/bootstrap/rc.local'
Source file not found: '/media/mmcblk0p1/bootstrap/rcS'
Source file not found: '/media/mmcblk0p1/bootstrap/fang_hacks.sh'
Source file not found: '/media/mmcblk0p1/bootstrap/fang_hacks.cfg'

/media # ls -lia
    143 drwxrwxrwt    6 root     root           120 Jan 28 09:03 .
     76 drwxr-xr-x    1 1000     232            272 Jan  1  1970 ..
      1 drwxrwxrwx    5 root     root           512 Jan  1  1970 mmcblk0p1
   1391 drwxrwxrwx    2 root     root            40 Jan 28 09:03 mmcblk1
      1 drwxrwxrwx    5 root     root           512 Jan 28 09:03 mmcblk1p1
   1390 drwxrwxrwx    2 root     root            40 Jan 28 09:03 mmcblk1p2

I tried to reboot but nothing changes.

Where do I need to change the path to make it works ?

Thnks !

ngtwolf, 2017-01-28:
Been there.  Take the SD Card out.  Unplug it.  Plug it back in.  Wait 2 minutes til you see the blue lights on the camera and it stops flashing.  Put the SD Card back in.  If it doesn't work, try the above again..  If you just remove the SD and put it back in, it won't work since the paths get all messed up.

Vennerstrand, 2017-01-28:
I had three units where I just put in the sd-cart and powered up the camera and everything worked nicely. This morning, my other two refused to work. Only vfat partition p1 is mounted if booting with the sd-card. No web-acces either. I had to start the units without sd-card. Insert the card and wait for hammer sound. Web is now operational. Then enable rescue telnet. Telnet to the device, mount the p2 partition using 'mount /dev/mmcblk0p2 /media/mmcblk0p2'. THEN I can apply the hack.
Subsequent reboots seem to mount p2 without issues. Have not been able to figure out why.

samtap, 2017-02-02:
Auto-mounting of ext2 partitions is part of the hack. So you have to apply the hack first (inserting the sd-card AFTER the device has booted) but once applied you can leave it inserted and should mount automatically at boot. Auto-mount is managed by the software already on the device so if it doesn't happen, the hack is not to blame.

closed 2017-02-02 by samtap