samtap/fang-hacks #22: Video source in MJPEG possible?

uros76, 2017-01-29:
Hi. First of all great work with the hack. I managed to get it working through instructions and can access video source through VLC. 
I use zoneminder on ubuntu server and having big challenges with RTSP feed through Ffmpeg source type. Picture is choppy, signal drops and so on. 

Any chance for a script that enables MJPEG video source instead of RTSP? I have much better results on zoneminder using MJPEG source from other cameras.

samtap, 2017-01-29:
You can use MJPEG by passing the -j parameter to snx_rtsp_server. Use --help for more options. You'll have to edit the script in /media/mmcblk0p2/data/etc/scripts to start the RTSP server with different parameters.

no1knows, 2017-01-29:
I'm using the default RTSP server with Zoneminder without issue. Make sure you leave all of the FPS settings blank in the Monitor settings for the cameras:

[image]

uros76, 2017-01-29:
Thanks for quick responses. I played with -j parameter and cant get it to work on zoneminder, works with vlc player though. 

no1knows, can you tell me what are you using as "source type" and "remote method" on zoneminder camera setup?

no1knows, 2017-01-29:
Sure:
[image]

[image]

I'm using the rtsp server that was compiled a while ago - before samtap created the fang-hacks. Not sure if it's changed.

federic0, 2017-01-29:
Same setup as @no1knows works for me with zoneminder (I have "24 bit colour" instead)

uros76, 2017-01-29:
I had same settings. BTW which version of ZM server are you using? I have 1.29.0

It does work now with decreased resolution and bitrate, manually set on camera script. 
Will monitor closely for anomalies. 
Thanks again.

closed 2017-02-17 by samtap