samtap/fang-hacks #23: Hack not working

Devangel67, 2017-01-29:
I write the image using win32 disk imager and boot the device without the sd-card then i insert the sd but nothing work, i get always the same "404" and "OpenDir faile".

I tried everything but nothing work..What are ami doing wrong ?
I use the last firmware.

PS: Sorry for my english.

laucho, 2017-01-29:
hi 
go the same thing, i put the sd in when the camera is powered.
A sound happened but the url is still 404, i tried it several times powered or not powered.

Thanks

laucho, 2017-01-29:
Ok it goes on :)
the url is http://YOUR_IP/cgi-bin/status (little mistake on the front page)
but on apply : 
`Applying fang hacks scripts...
Source file not found: '/media/mmcblk0p1/bootstrap/sdcard'
Source file not found: '/media/mmcblk0p1/bootstrap/rc.local'
Source file not found: '/media/mmcblk0p1/bootstrap/rcS'
Source file not found: '/media/mmcblk0p1/bootstrap/fang_hacks.sh'
Source file not found: '/media/mmcblk0p1/bootstrap/fang_hacks.cfg'`

Vennerstrand, 2017-01-29:
@laucho See my last post in #21. You will need to enable telnet from web, telent, mount sd-card partition p2 and then you can apply the hack.

Devangel67, 2017-01-29:
I don't have any sound, if i go to http://myip/ the page is empty but i don't get any ''404'' or 'opendir faile'. Well i think my camera is broken...Ahaha

Gilgameshismist, 2017-01-29:
Devangel67 That is a bit sparse, do you have more information?
What FW are you on? Which modifications have you made? Which versions of hacks are you using? What action have been performed before the camera stopped working? Is there an SD card in the device? If so: what is on the SD card? What is the status of the led? Does the IR sensor click if you cover the front of the camera? What happens if you push the Reset button?

Devangel67, 2017-01-29:
Well i unplug the camera, plug in and press the setup and everything works again. But the hack don't work, I'm use the latest firmware, no modification , i just used win32 disk imager to write the image on the sd card, so i get a fa32 partition and a ext2 partition and insert the card after the boot but nothing, 

I hear the sensor when i cover the front of the camera and mi home work well. Where is the reset button ?

Gilgameshismist, 2017-01-29:
The reset button is on the bottom of the camera, you used it to set up the WIFI connection. 
But what I hear now is that the camera works, but the hacks don't apply.
If you let the camera boot and the LED is blue you can connect with mi home, this is good!
Now you can pop in the prepared SD card, it will take a few seconds but you should hear a dingding sound. If not something is wrong with the SD (or the way the camera is mounting the sd).
If you wait some time after inserting the SD you should be able to go to camip/cgi-bin/status , does this load? (you need to be on the same wireless network as your cam to be able to reach that page!)

Devangel67, 2017-01-29:
I inserted in the camera but no sound, always error 404 but the camera saved pictures on the sd card...I'm gonna try another sd card to see if something change, its a sandisk ultra 16gb.

Gilgameshismist, 2017-01-29:
If the camera doesn't give the dingding sound after the SD is inserted, it probably doesn't detect it as SD. Either the SD is corrupt, or the image wasn't written to the SD right. (Which I can't help you with, I never used the image file, I made the partitions using gparted)

Devangel67, 2017-01-30:
This time i have the status page with this : 
README.md
bootstrap
snx_autorun.sh
time_lapse
record

But the status page said 404...Better than before i guess.

Gilgameshismist, 2017-01-30:
I don't think that is the status page, that looks like the result of hello.cgi
it looks like snx_autorun.sh doesn't run. 
Which version of diskimage are you using? If you have firmware v3.0.x you should use image v0.2.0.

Devangel67, 2017-01-30:
Sorry, that the hello.cgi page, i used gparted to create a fat32(i am wrong)partition and ext2 and copied the files from github.  The camera save files on the sd card but i don't understand why the snx_autorun doesn't run... 

I tested a bit and the camera doesn't detect the sd when i let the camera boot, only if i insert the sd and plug in the camera... Thank you anyway :).

Gilgameshismist, 2017-01-31:
Hi @Devangel67, any progress? Did you manage to get it running?
Did you try another SD card?
 Regards,

Devangel67, 2017-02-01:
I tried some sd cards, same problem...Telnet > refused, 192.168.15.18/cgi-bin/ > 403, 192.168.15.18/cgi-bin/status > 404.

Mi Home detect the sd card and i can save picture or something like that but i don't undersand why the script doesn't run..

Something is wrong with my camera i'm sure but why everything work perfectly in mi home ?...

The only  strange thing in mi home, if i boot with the sd card no problems but if i insert the sd after the boot nothing is detected.

I need a miracle :D

tooviral, 2017-02-01:
Have you tried different sd cards??

Devangel67, 2017-02-02:
Of course i tried but same results. I am going to buy another one to see if this one is faulty.

samtap, 2017-02-02:
To test if mounting works correctly, insert the sdcard AFTER the device has booted. Then go to http://device-ip/cgi-bin/hello.cgi. This is a default script, not part of the hacks, and should list the contents of the vfat partition (including snx_autorun.sh). If you get some error (openDir failed), you may have a corrupt filesystem that prevents it from getting mounted. Try to repair it or use a new card and start from scratch.

Devangel67, 2017-02-04:
I tried many sd cards, nothing work...When i boot with the sd card i get the hello.cgi but only in this case. What can i do now ?

Devangel67, 2017-02-05:
Finally the hack work but in the "manage script" the ftpd, rtsp-server, ir-control are "NOK".

samtap, 2017-02-06:
@Devangel67 Please read through #17 and the [FAQ](https://github.com/samtap/fang-hacks/wiki/F.A.Q.) or maybe you can find some help on discord chat?

Devangel67, 2017-02-08:
Thank you very much, everything work yet :).

closed 2017-02-08 by samtap

iliya2003, 2017-08-10:
The same problem, how did you fix the 404 problem??