samtap/fang-hacks #26: snx_rtsp_server crashing with standard parameters

conradhagemans, 2017-02-05:
Thanks for the great hack.
I found out that on my camera the rtsp server crashes all the time.

default in script: snx_rtsp_server -W 1920 -H 1080 -Q 10 -b 4096 -a >$LOG 2>&1 &

I changed it to use the defaults of the server:

snx_rtsp_server >$LOG 2>&1 &

Now it runs fine stable. But with lower resolution.

Acharyapr, 2017-02-09:
Having same prob here.. after starting rtsp server, camera is loosing network connectivity.. i think so camera hardware is not capable for full HD but interpolated to 1080P

Pierinopower, 2017-05-03:
inthis mode (snx_rtsp_server >$LOG 2>&1 &) Works fine, no crash! Thank you

yuyi, 2017-08-26:
Hello,

I found out that is about the **-a** parameter. (enable audio output)

### unstable
```
snx_rtsp_server -W 1920 -H 1080 -Q 10 -b 5000 -a >$LOG 2>&1 &
snx_rtsp_server -W 1280 -H 720 -Q 10 -b 2048 -a >$LOG 2>&1 &
snx_rtsp_server -W 1920 -H 1080 -Q 10 -b 2048 -a >$LOG 2>&1 &
```

### stable
```
snx_rtsp_server >$LOG 2>&1 &
snx_rtsp_server -W 1920 -H 1080 -Q 10 -b 5000 >$LOG 2>&1 &
```

After I deleted the **-a** parameter, the fang can run stably at 1080p resolution.
You can try it :p