samtap/fang-hacks #28: Help needed to setup development invironment

conradhagemans, 2017-02-07:
I have a fully working hacked camera. Thanks for the SD-Image.
I setup an development invironment with the 1.50 SDK.
I can compile the RTSP-Server. But this version gives errors on the device.

snx_codec_rc_qp:1149: ioctl: Invalid argument
snx_codec_rc_qp:1149: ioctl: Invalid argument
open /proc/isp/md/block-reports failed.
open /proc/isp/md/block-reports failed.
open /proc/isp/md/block-reports failed.
open /proc/isp/md/block-reports failed.

So I think I have something wrong setup in the SDK. I guess the wrong model.

Who compiled the RTSP-Server that is included in the SD-Image?

samtap, 2017-02-08:
I cannot help you with the SDK

closed 2017-02-08 by samtap

gitingear, 2017-04-25:
have encountered this error as well, does anyone have any suggestion on how to solve it, thanks