samtap/fang-hacks #32: Bad video quality

otbe, 2017-02-15:
Hi everyone,

I use your hacks on my Xiaomi 1080p cube cam (iSmartAlarm) and it works pretty well, but the video quality is quite bad compared to the original MiHome app. Is there anything I can do about it?

Thank you!

samtap, 2017-02-15:
You can increase the bitrate of the rtsp stream but the default works fine for me. In what way is the quality bad? Stuttering and/or artifacts can be caused by a bad wifi signal... in that case a lower bitrate will improve quality. Or increased cache size of client, it all depends on your situation.

otbe, 2017-02-15:
I get a lot artifacts. Its somehow blurred after 0.5-1m distance. I don't see this artifacts/blurred video with the original MiHome app. Same place, same Wifi. I use VLC to view the rtsp stream (MacBook) and Synology's DS cam app (iPhone) with same result on both devices/clients.

I guess you mean increase the value of ```-b``` in ```snx_rtsp_server -W 1920 -H 1080 -Q 10 -b 4096 -a >$LOG 2>&1 &```? Will try that in the next days...

samtap, 2017-02-15:
Yes -b is for the bitrate. The cloud app has three quality settings and a variable bitrate. Even in the highest setting the bitrate doesn't exceed ~2Mbit. I've only viewed it on my small phone screen but the image looks pixelated so they may even be using the low resolution substream (640p?). RTSP stream works much better for me, crystal clear on a 50" tv. But it does take a toll on wifi bandwith...

otbe, 2017-02-15:
Mh, tried several options. Higher & lower bitrates, higher & lower resolutions. Result is always the same: (crystal) clear within 0.5m, blurred beyond. Im a bit helpless :/

Gilgameshismist, 2017-02-15:
Sounds like you have a unit which has a focus problem. I am not aware whether you can manually focus the lens.

otbe, 2017-02-15:
That's what I thought too, but why its perfect with the original cloud app? Makes no sense to me :/

psmanek, 2017-02-20:
Yes, video quality is quite poor. It's good for bandwidth, but it would be nice to set HD quality. I don't know if this is rtsp server settings or camera itself. Even if i set higher bitrate quality is not enough.

Acharyapr, 2017-02-20:
First of all thanks samtap for creating this "Hacks". I just purchased this camera from Gearbest @ 23$.. its very cheap for ip camra....but its not a Generic IP Network Camera, Everything is locked by firmware...If you cant fully access this camera from pc (for Big Display) or record stream in NVR this is a problem of restrictions... Samtap gave wings to this camera to work as a Generic Network IP Camera... If anyone is not getting good performance, its a problem of Network or PC, Originally this is a Ihealthcare's Spot Camera with Hardware capability of 1280X720 Video only.

psmanek, 2017-02-20:
Yes, Big thanks to @samtap for his work.

@Acharyapr, only 1280x720? So full HD is only by resizing image? That's no good ;/

Thanks everyone

Acharyapr, 2017-02-21:
Playing with different resolution since couple of hours.. Does anyone tested it with more than 24P at any resolution ?

ant-thomas, 2017-03-02:
For those interested in increasing the video quality I recommend increasing the bitrate in the `snx_rtsp_server` script. I've tried 8192 and 16384. There is a huge difference in quality. I've found that in the dark 4096 doesn't actually give a bitrate over 1mbps - more like 0.1mbps, whereas using 16384 provides a bitrate of over 5mbps at times, and therefore a much higher quality image.

Zmove, 2017-03-08:
Could you please explain how do you change that scripts ?

Thanks

Acharyapr, 2017-03-10:
Just ssh the cam and edit it by vi editor.

eterpstra, 2017-03-21:
Can you tell me were the script is located?

iothing, 2017-03-21:
/media/mmcblk0p2/data/etc/scripts/20-rtsp-server

eterpstra, 2017-03-22:
Oke thanks I will try this tomorrow.

I have received the camera 2 days ago. Everything is working fine. However, what is the average bitrate for this camera when using the Mihome app? I am only getting 30 - 200 kbs/sec with this cam.  Is this normal?

sssangel, 2017-04-06:
I compare with my xiaomi Yi Cam 720p on RTSP, the quality is very bad (same wifi, same place).
I tried increase the -b to 8192 and 16384 but I have a lot of artifacts and the video lag.

ant-thomas, 2017-04-06:
I've found a good balance of quality and speed to be "5000". Lag and artifacts are probably caused by poor wifi. Which could be that the camera wifi antenna isn't so great compared to the Yi Cam

sssangel, 2017-04-06:
Thank you, I will try tonight.
I tested all both cam near the router (10cm lol)

TTome123, 2017-04-14:
For me changing to: snx_rtsp_server >$LOG 2>&1 &

worked perfectly. Previously changing res/bitrate was not enough.

eterpstra, 2017-04-15:
What do you mean with $LOG 2>&1 &?

w3ssisback, 2017-05-22:
Same issue with the quality of rtsp vs the cloud app.

Tried increasing the bitrate but that didn't make a difference. The image appears to be zoomed in when I do this. Anything else I can try?

DavidRayner, 2017-05-22:
You could maybe test in a location with better WiFi reception

w3ssisback, 2017-05-23:
Wi-Fi reception is ok. I moved it in the same room as the router and same results. Cloud app looks really good though

aminpi, 2017-12-29:
i have the same bad video quality... 
is it possible cause because i have 4gb sdcard and FAT(not FAT32) file system on it?

cslinmiso, 2021-12-29:
got the lag issue