samtap/fang-hacks #34: Cannot run RTSP server without cloud apps

Cr4z33, 2017-02-18:
First let me ask if the cloud apps are still the Xiaomi's ones? 

Second if I disable them I can't run the local RTSP server successfully. 
I always get the NOOK message. 

I am running the camera Wi-Fi into client mode and using your latest hack.

samtap, 2017-02-18:
You must be mistaken, you can only run RTSP server *without* the cloud apps

Cr4z33, 2017-02-18:
That's what I was trying. 

However game is over as the device now is stuck with a static orange LED. :(
All this happened after a reboot. 

Any chance to fix somehow it with a serial connection?

samtap, 2017-02-18:
There's nothing wrong with a static orange led, it just means the cloud apps are disabled. Instructions to recover from a broken wifi config are in #17

Cr4z33, 2017-02-18:
Thank you I am gonna try ASAP! :D

psmanek, 2017-02-20:
> You must be mistaken, you can only run RTSP server without the cloud apps
Really? On my camera RTSP works with cloud apps ;/

samtap, 2017-02-20:
You must have a magical camera! I would put it on ebay for at least 1000 bucks since you have the only camera in the world that can magically share the hardware between cloud apps and rtsp server.

psmanek, 2017-02-20:
no! You misunderstood me. Cloud apps does not work, but in settings camera is in Cloud apps mode (RTSP enabled).

ephestione, 2017-02-28:
So, to get it straight, what are the advantages of having wifi client mode instead of cloud apps mode? Just that cloud apps are not running, so the camera boots faster? Otherwise there is no privacy drawback whatsoever with having cloud apps enabled when hacks are enabled?

samtap, 2017-02-28:
Why don't you email Xiaomi and ask them about privacy drawbacks when running the cloud apps for a short while and then killing them? How should I know?

I can only guess, and I guess the cloud apps connect to China immediately after the WiFi is up. In my case that is not acceptable but everyone is free to decide on their own.

samtap, 2017-02-28:
What are you doing to put it in wifi client mode?

ephestione, 2017-03-01:
I MIGHT have nailed it. I tried both pre-setting the cfg files (fang hacks and wpa supplicant), and also setting them from the network tab, but after a reboot I always got the static orange led.
This time I tried once again by manually setting the cfg files, resized the 64gb card, and now when I boot, I get the flashing blue alternating with orange (so cloud apps should be disabled), and when I open the network tab of the hacks page, I get, under the wifi client radio button section, a "connected" notification, so it should mean I did connect with wifi standalone without cloud apps.

samtap, 2017-03-01:
Great you might have nailed it, I'm happy for you. Though I've no idea where're you're basing your info on, it's quite amusing...  I'm going to ask you to stop posting here, as other people might find it and think the info is correct. This area is reserved for reporting actual issues. If you have questions please go to discord.

closed 2017-03-01 by samtap