samtap/fang-hacks #39: No image URL available

joostjdj, 2017-02-27:
First off all, thank you very much for this hack. It works wonderfull.

Can a image URL be added? This to use it for snapshots for programs and applications that don't support RTSP?

ant-thomas, 2017-02-28:
Probably won't work for you if there's a lack of RTSP support in the software you want to use but FFMPEG can capture a single frame from an RTSP feed.

empierre, 2017-03-01:
as said elsewhere this could do it: 
/usr/bin/avconv -i rtsp://192.168.0.29:554/unicast -frames:v 1 -an -y /tmp/snapshot.jpg

maybe aconv could be added to the image with a cgi wrapping ? would be much useful sor systems with domoticz ?

ant-thomas, 2017-03-01:
The image currently contains a build of ffmpeg which can be used if
required.

joostjdj, 2017-03-06:
What is the image URL to use for example in Domoticz for ffmpeg?

MarkG55, 2017-03-12:
Another alternative is to just use my modified version of the snx_rtsp_server:

https://github.com/fritz-smh/yi-hack/issues/118#issuecomment-277503743

deennoo, 2017-03-22:
@MarkG55  your suggestion works great thank you

i made a full how to in french with a .img file 

you can get it here : http://domo-attitude.fr/camera-xiaomi-xiaofang-flux-jpg-domoticz-jeedom/

theo-69, 2017-06-17:
@MarkG55  and @deennoo  only issue now is the rtsp stream is not working anymore, how can I use both?