samtap/fang-hacks #41: OpenDir failed - hello.cgi

tarat73, 2017-02-28:
Hello,

Thank you for your effort.
I have got this camera yesterday from Gearbest.
I connected with MiHome, but it is a security risk.
I would like to install your solution.
I have created a Pendrive by your description and image (V 0.2). I checked the drive with Linux and it seems like correct.
I booted the camera and after the blue constant light I inserted the PenDrive.
The message from http://192.168.1.33/cgi-bin/hello.cgi -> OpenDir faile
The message from http://192.168.1.33/cgi-bin/status: 404 Not Found
Could you help me?

Thanks,

Attila

ephestione, 2017-02-28:
You are supposed to use an sd card, not a pendrive

tarat73, 2017-02-28:
Thank you.
I inserted the SD card but the problem is te same as I described before :(
Do you have any other idea?

tarat73, 2017-02-28:
OK. Success.
I restarted the camera and the next restart fixed the problem. Sorry

closed 2017-02-28 by tarat73