samtap/fang-hacks #42: Showing date/time on rtsp stream?

ephestione, 2017-02-28:
Happy user of two hacked xiaoyi ants (another one is still in the drawer), I like the clock on the video feed, so at least I know the stream is going and not frozen.
Is there a way to have it on this camera as well?

samtap, 2017-02-28:
Yes, see WIKI: https://github.com/samtap/fang-hacks/wiki/Controlling-the-text-overlay

closed 2017-02-28 by samtap

madmax2, 2021-05-01:
@samtap can you please add more step by step instructions in Wiki https://github.com/samtap/fang-hacks/wiki/Controlling-the-text-overlay for newbies to follow?
Jumping to another issue tracker https://github.com/fritz-smh/yi-hack/issues/118#issuecomment-273896219, which also does not have clear steps to follow is not easy for newbies.

e.g. where do you put  snx_isp_ctl  etc?