samtap/fang-hacks #43: feature request: motion detection

empierre, 2017-03-01:
Hello,

  I'm not sure if the motion detections is being used there ? 

  There's been a lot of discussion around this here it seems:
https://github.com/fritz-smh/yi-hack/issues/118

samtap, 2017-03-01:
Yes motion detection has been mentioned before and is on my wish list. 
Can you provide an explanation how you would expect it to work? Perhaps use the wiki (there's a list of feature ideas). 
Motion detection is not usually a feature provided by an rtsp-server, but with the given hardware I don't think we have any other option. Only one process can use the camera hardware at the same time, so motion detection code must be integrated in the same executable as the rtsp server.
Minimum requirement would be some kind of ring-buffer. Since when motion is detected you usually want a couple seconds of video from before the motion as well. Then perhaps a configurable amount of time during which video is recorded to file and motion triggers are ignored.
Note you can already play with motion detection in the current image with the snx_isp_md binary (testing different sensitivity settings etc.) if you stop the rtsp server.

empierre, 2017-03-01:
Hello, based on the foscam API I currently use:
- alarm on motion detection (from API call I use the on/off, and trigger the schedule from domoticz/google calendar)
- in case of movement during the alarm period, on motion detection, it takes a snapshot and send it by email, and record video with pre-buffered images your mentioned for a configurable perio of time (I set it to 5 mn).

magnets110, 2017-03-02:
snx_isp_ctl is able to pull events while the RTSP server is running. line #266

`void snx_md_report_thread(void *arg)`

tarat73, 2017-03-02:
@magnets110 :+1: Could you describe the solution?
Do I have to modify the /data/etc/scripts/20-rtsp-server?
start()
{
  LOG=/dev/null
  echo "Starting RTSP server..."
  snx_rtsp_server -W 1920 -H 1080 -Q 10 -b 4096 -a >$LOG 2>&1 &
**snx_isp_ctl  --mdset-en 1**
  echo "$!" > "$PIDFILE"
}

and where do I have to use the void snx_md_report_thread(void *arg) ?

Thank you for your help...

studioBeaver, 2017-03-15:
I tried snx_isp_ctl --mdset-en 1, but not much changed. I think some thing else is missing.
I know it's not easy to implement motion detection. I'm already using this camera as AP, with only FTP and telent enabled, so the camera is recording all the time to an SD card. But the problem is that there is no way to detect if there was any movement. So if someone could make some kind of log file, when motion is detected, that would be great! If here could be only whe time of an event in the log file, that would be enough for me. So I can at least check the files that were recorded in that time. Just an idea, if this could be easier to implement.
Or if someone has made some other solution. thanks

iothing, 2017-03-20:
> so the camera is recording all the time to an SD card

What program are you using for this? ffmpeg? Would you mind posting the commandline?

corollas, 2017-03-22:
the older Yi camera had motion detection recording capability with loop recording saved onto SD and accessible HTTP server
https://github.com/fritz-smh/yi-hack

this camera hardware not capable?

iothing, 2017-03-28:
First we would need to modify snx_rtsp_server to save the h264 stream to the SD card, optimally by putting it in a mp4 container so that it can be easily played back by vlc and mplayer without specifying type, resolution and fps.

For the motion detection part there are two general ways of doing it:
 1. we get access to the h264 hardware encoding motion estimation block to detect significant changes in the image
2. we use the open source software motion, however this won't work at full HD, so maybe it's possible to use a low resolution substream that the hardware encoder provides while writing the fullhd stream into a circular buffer and triggering a copy of it to individual files at motion's request. I think motion does not support h264 video streams so we need to re-encode to mjpg the substream.

studioBeaver, 2017-04-03:
@iothing 
"What program are you using for this? ffmpeg? Would you mind posting the commandline?"
I think this is standard camera functionality. What I did, was first disable all the running scripts, and only leave FTP and telnet script active, so I can access the SD card via FTP. And I'm using the camera as AP, so I connect directly to the camera and with FTP I can browse the SD card. This way, the camera is recording videos all the time. They are saved on FAT32 partition in recordings folder, and then every folder has the name of the current hour and the video file has the name of the minute that was recorded. The videos are split every minute. I have resized the FAT32 partition to use the max space for the video files.
But as you can guess, this is pretty hard to use for some kind of motion detection... It's too big to check all the files that were recorded in a day... If there would be some kind of way to trigger something when any motion would be detected, it could at least write a log with the timestamp of such event.
Oh, and one more thing. The reason I had to disable all other scripts is that only one process can use the camera, so RTSP has to be disabeled.
I will try motioneyeOS on raspberry, since I read that it has some motion detection...
I hope there will be some more development for this nice camera...

studioBeaver, 2017-04-19:
If anyone is interested, I manged to get this camera working with https://github.com/ccrisan/motioneyeos
Still testing it, but for now it's working as I wanted with RTSP stream, and it's detecting motion. I'm running the MotionEyeOS on a raspberry PI3. The setup was fairly easy.

desidude2000, 2017-05-01:
I would be interested.  i  have quite a few armel devices setup (running debian) on my home network with native compile.   Buffalo NAS systems and pogoplug (for those that remember it).    In any case, please let me know how i can get started with motioneyeos.    Motion detection would be very useful.

niczoom, 2017-05-05:
Sounds interesting! I currently use tinyCam PRO on an android phone to detect motion and record video, works well enough, depending on the android hardware though. I think it would better/preferable to have the camera hardware do all the work, if possible?

@samtap Any progress or updates on motion detection integration?

seblucas, 2017-05-30:
Sorry for the questions but my camera is still in transit.

Does the camera has a real hardware motion sensor (PIR) ? it seems so by reading the description but I'm less interested in software motion sensor. If you we can record the latest 5 second of video in a ring buffer and wait for motion to properly save it elsewhere.

Anyway thanks a lot for this repo and all the discussions around it.

jimmyktp, 2017-10-04:
@seblucas , don't think there's real hardware motion sensor PIR. Refer to the teardown --> http://www.shoudian.org/thread-486708-1-1.html

seems like the original iSpotAlarm has a PIR, but not Xiaofang. seems like some cost cutting measures here. I wonder if this thread -- > https://github.com/samtap/fang-hacks/issues/193 we can add a PIR and enable the hardware motion sensor.

dvv, 2018-01-13:
A skeleton for motion detection: https://github.com/samtap/fang-hacks/wiki/WIP:-Motion-detection

adrianmihalko, 2018-03-24:
@dvv what is WIP and where to put this file? Thank you.

dvv, 2018-03-26:
@adrianmihalko WIP is for Work-In-Progress. This file is to be run on the camera.

g90210, 2018-04-26:
@dvv how exactly this script has to be used? Do you need to add it to cron? How accurately the script is working so far? Thanks

dvv, 2018-04-28:
@g90210 just run it on the cam and ensure it won't terminate so no need in cron. I'm not using it since I don't need motion detection feature where the cam right now. You'd play with parameters of `snx_isp_ctl` and "highlylikely" these parameters should depend on day/night mode.

bhaap, 2018-11-27:
So I've made a mqtt motion detection based of your code @dvv (thanks!)

First you will need mosquitto_pub, snx_isp_ctl and snx_isp_md in your /media/mmcblk0p2/data/usr/bin folder. They all need to be executable (run chmod 775 * in that folder after the binaries are in place)

Then you need to create two scripts.

This one is located in /media/mmcblk0p2/data/etc/scripts
22-mqtt-motion:
```sh
#!/bin/sh
PIDFILE="/var/run/rtsp-md.pid"

status()
{
  pid="$(cat "$PIDFILE" 2>/dev/null)"
  if [ "$pid" ]; then
    kill -0 "$pid" >/dev/null && echo "PID: $pid" || return 1
  fi
}

start()
{
  echo "Starting RTSP motion detection..."
  rtsp-md.sh /dev/null 2>&1 &
  echo "$!" > "$PIDFILE"
}

stop()
{
  pid="$(cat "$PIDFILE" 2>/dev/null)"
  if [ "$pid" ]; then
     kill $pid ||  rm "$PIDFILE"
  fi
  pids=$(ps w | grep Detected | grep -v 'grep' | awk '{print $1}')
  echo $pids
  for pid in $pids ; do
    kill -9 $pid
  done
  pids2=$(ps w | grep snx_isp_md | grep -v 'grep' | awk '{print $1}')
  echo $pids2
  for pid in $pids2 ; do
    kill -9 $pid
  done
}

if [ $# -eq 0 ]; then
  start
else
  case $1 in start|stop|status)
    $1
    ;;
  esac
fi
```

And this one is located in /media/mmcblk0p2/data/usr/bin (run chmod 775 * after creation of this file to make it executable)
rtsp-md.sh:
```sh
#!/bin/sh

# Copyright 2018 Vladimir Dronnikov
# GPL
# MQTT added by Johan Leirnes 2018-11-28

snx_isp_ctl --mdset-thre 1000 # YMMV
snx_isp_md 2>&1 | awk '
/^Detected Moving!$/ {
  if (alarm == 0) {
    alarm = 1
    print "+alarm"
    system("mosquitto_pub -h 192.168.1.xx -p 1883 -t /cam/pir -m 1 -u USERNAME -P PASSWORD")
  }
}
/(have no motion status = 0)/{
  if (alarm != 0) {
    alarm = 0
    print "-alarm"
    system("mosquitto_pub -h 192.168.1.xx -p 1883 -t /cam/pir -m 0 -u USERNAME -P PASSWORD")
  }
}
'
```

-h is the mqtt server host adress
-t is the topic on which the message will post
-u is the username for the mqtt server
-P is the password for the mqtt server.

Hope this helps someone.

bhaap, 2018-11-27:
Im using the motion detection to trigger my CCTV to connect to the camera and start recording for a couple of minutes/ when the motion stops.
This is so that i do not have alot of bandwidth usage when there is no need for it.

dvv, 2018-11-28:
\o/

bhaap, 2018-11-28:
I've noticed that snx_isp_md only works for lower resolutions.

im using this:
snx_rtsp_server -W 1280 -H 720 -F 15 -Q 10 -b 2048 >$LOG 2>&1 &

divemasterjm, 2020-03-14:
bhaap, can you please tell me  where to get mosquitto_pub, snx_isp_ctl and snx_isp_md? i want to try your mqtt workaround please

divemasterjm, 2020-03-14:
> I've noticed that snx_isp_md only works for lower resolutions.
> 
> im using this:
> snx_rtsp_server -W 1280 -H 720 -F 15 -Q 10 -b 2048 >$LOG 2>&1 &

where i can get snx* and mosquitto binaries for xiaofang?
thanks

dvv, 2020-03-14:
@divemasterjm #305

agujensen, 2021-02-08:
The status of the script is in red, NOK, what can it be?