samtap/fang-hacks #45: MJPEG Stream

PJF16, 2017-03-01:
Would it be possible to add a MJPEG Stream? So we could stream the picture live to a website.

ant-thomas, 2017-03-02:
It is already possible
https://github.com/samtap/fang-hacks/issues/22

PJF16, 2017-03-02:
Thank you for your answer!

Adding -j as a parameter in /etc/scripts/20-rtsp-server unfortunately doesn't work. 

`snx_rtsp_server -j -W 1920 -H 1080 -Q 10 -b 4096 -a >$LOG 2>&1 &`

ant-thomas, 2017-03-02:
Try this instead

`snx_rtsp_server -j 60 -W 1920 -H 1080 -a >$LOG 2>&1 &`

PJF16, 2017-03-02:
Now I get only a red "NOK" in the webinterface.

ant-thomas, 2017-03-02:
Can you ssh in and run the command manually?

`killall snx_rtsp_server` should kill any existing instance
`snx_rtsp_server -j 60 -W 1920 -H 1080 -a` should let you run the server manually

I found that sometimes I got the error
`Failed to create RTSP server: bind() error (port number: 554): Address already in use `
But waiting and trying again eventually released it

PJF16, 2017-03-02:
Yey! That works!

But it seems like I don't get a legit MJPEG stream over http. Only over rtsp.

ant-thomas, 2017-03-02:
I assume you've seen this:
```
snx_rtsp_serverVersion:V00.01.06p
Usage :
	 snx_rtsp_server [-a] [-j mjpeg_qp] [-m] [-P RTSP port][-T RTSP/HTTP port][-Q queueSize] [-M groupaddress] [-b bitrate] [-W width] [-H height] [-F fps] [-i isp_fps] [device]
	 -Q length: Number of frame queue  (default 10)
	 RTSP options :
	 -u url     : unicast url (default unicast)
	 -m url     : multicast url (default multicast)
	 -M addr    : multicast group   (default is a random address)
	 -P port    : RTSP port (default 554)
	 -T port    : RTSP over HTTP port (default 0)
	 V4L2 options :
	 -F fps     : V4L2 capture framerate (default 30)
	 -i isp_fps : ISP capture framerate (default 30)
	 -W width   : V4L2 capture width (default 1280)
	 -H height  : V4L2 capture height (default 720)
	 V4L2 H264 options :
	 -b bitrate : V4L2 capture bitrate kbps(default 1024 kbps)
	 -g gop     : V4L2 capture gop (default 30 )
	 device     : V4L2 capture device (default /dev/video1)
	 V4L2 MJPEG options :
	 -j mjpeg_qp : MJPEG streaming and qp (default is 60)
	 -c capture path enable (default is disable)
	 -s encoder scale (1/2/4) (default is 1)
	 -a         : enable A-law pcm streaming 
	 H264 example : snx_rtsp_server -a -Q 5 -u media/stream1 -P 554
	 MJPEG example : snx_rtsp_server -W 640 -H 480 -j 120 -Q 5 -u media/stream1 -P 554
```
The obvious way to me would be to use `-T port` but that doesn't seem to work for me.

PJF16, 2017-03-02:
Yes. I tried snx_rtsp_server -j 60 -T 8082 -W 1920 -H 1080 -a

Doesn't work for me either. :(

PJF16, 2017-03-02:
My next idea would be to make a stream with ffmpeg.

PJF16, 2017-03-02:
I got the ffserver running but ffmpeg said when I wanted to feed the server, that the outputfile (I used /dev/video2) doesn't contain any stream...

ant-thomas, 2017-03-02:
Try /dev/video1 - that's what snx_rtsp_server uses

ant-thomas, 2017-03-02:
Or try ffmpeg/ffserver using the rtsp address as the input.

PJF16, 2017-03-03:
I thought, that snx_rtsp_server uses video2. But I also tried /dev/video1 - same result. 

It's also possible, that the ffserver is just wrong.

ksprelude01, 2017-06-07:
Has anyone gotten the MJPEG stream to work correctly yet...I was able to get the server running but can not find output stream.

j2innet, 2020-04-27:
make sure process is clean
check by
ps | grep snx_rtsp_server

if found, then run
killall snx_rtsp_server

try command with new parameters

check help
snx_rtsp_server -help

don't run
snx_rtsp_server -j -W 1920 -H 1080 -Q 10 -b 4096 -a >$LOG 2>&1 &

run
snx_rtsp_server -W 1920 -H 1080 -Q 10 -b 4096 -a >/dev/null 2>&1 &