samtap/fang-hacks #48: Problem running the RTSP

TamerLanTK, 2017-03-05:
Hey. I have a problem running the RTSP.  How to fix the problem?
[image]

simoneluconi, 2017-03-05:
Hi! it's the problem that i had. You need to disable the cloud app first (run 00-stop-cloud) and then run the rtsp-server. But you shouldn't have the network error :/ Maybe you also have misconfigured some network configuration.

TamerLanTK, 2017-03-05:
when I start the script (00-stop-cloud), I get the message
`Running script '00-stop-cloud'...
/media/mmcblk0p2/data/etc/scripts/00-stop-cloud: source: line 3: can't open '/etc/fang_hacks.cfg'`

samtap, 2017-03-07:
I've no idea how you ended up in this situation... If you remember please list all steps leading up to this. You did click the "apply" button on the status page to enable the hacks, right?

Spartacus-6, 2017-03-08:
I'm stuck ???

Ran stop script - Running script '00-stop-cloud' but can't disable cloud ???

[image]

Spartacus-6, 2017-03-08:
[image]

samtap, 2017-03-08:
This isn't going anywhere. If you really think there's a bug, open a new issues and
**include a thorough description of your problem and the steps you've taken to get there**

If you're just looking for support because you're in over your head, go to discord and ask for help.

closed 2017-03-08 by samtap