samtap/fang-hacks #49: Alternative text editor (nano)

ant-thomas, 2017-03-07:
Would it be possible to add an alternative (and more user friendly) text editor?
My personal preference would be nano, assuming this is trivial to do?
Main reason would be so it is a little easier to create and edit files on the camera over SSH whilst running. Obviously this can be done with vi

samtap, 2017-03-08:
Actually I've been thinking about including a js text-editor for editing scripts from the web interface...

Meanwhile, [here's](https://www.patmas.net/nextcloud/index.php/s/GLMfWgie7XdzwAx) a nano build. Extract to /media/mmcblk0p2/data, you'll also have to set TERMINFO=/media/mmcblk0p2/data/usr/share/terminfo (add it to /media/mmcblk0p2/data/etc/profile to make it stick).

Let me know how it works out, I'll add it to the next image

ant-thomas, 2017-03-09:
Thanks.
Extracted and copied to the correct place. Added the TERMINFO line to profile but I'm getting the following error
`Error opening terminal: xterm-256color.`

samtap, 2017-03-09:
That's related to terminfo not being found. What happens when you run "TERMINFO=/media/mmcblk0p2/data/usr/share/terminfo/ nano"?

ant-thomas, 2017-03-09:
That launches nano correctly.

samtap, 2017-03-09:
Did you add export in profile like this:
export TERMINFO="/media/mmcblk0p2/data/usr/share/terminfo"

ant-thomas, 2017-03-10:
Sorted, my error. Forgot to include "export". Working now, thanks!

closed 2017-03-10 by ant-thomas