samtap/fang-hacks #50: Nothing is mounted on /media/mmcblk0p2

sirmany, 2017-03-08:
I have writed the sd image on a 16gb microsd and booted ok and apllied de hack, i have extended the 2nd partition in the script menu. Now i have rebooted and in the scipt page says the message on the subject: "Nothing is mounted on /media/mmcblk0p2". if i get into the tlenet and mount manually on /media/mmcblk0p2 this show me a mount failure because is busy.

Thanks!

samtap, 2017-03-08:
The filesystem is resized during boot, it cannot be mounted until the resize is complete. You can tell by checking the process list on the status page, resize2fs will be running, just wait until it's finished.
It can take a long time (much longer then on a regular pc), i.e. for a 64gb card it takes at least an hour...

sirmany, 2017-03-08:
/media/mmcblk0p1/bootstrap/resize2fs -f /dev/mmcblk0 <-- This?

thanks for your fast answer! :)

pattyland, 2017-03-09:
This a crucial tip that should in the installation instructions! Or maybe in the message in the web frontend!

samtap, 2017-03-09:
Yes it should be on the status page more prominent, will add that in next release. I know of at least a couple users who were too impatient and rebooted during resize, which risks completely messing it up the sdcard :roll_eyes: 

It's mentioned on the release page though, it just turns out not everybody reads that. Go figure.

pattyland, 2017-03-09:
Put me on that list :disappointed: 
'Filesystem                Size      Used Available Use% Mounted on
/dev/mmcblk0p2          145.3M     53.1M     84.7M  39% /media/mmcblk0p2
'
How do trigger another resize try?

samtap, 2017-03-10:
If you're lucky it will retry when you reboot the device. Check /tmp/hacks.log to see if it works. The resize is triggered when a file exists (/etc/.resize_runonce), which is deleted when the resize succeeds.
If you're not so lucky the filesystem is damaged, you can try to repair with /media/mmcblk0p1/bootstrap/e2fsck. It will likely restore the original partition table. You can use the webif to update it, triggered when the file /media/mmcblk0p2/data/.resize exists.

Ofcourse you can also just write the image again and start from scratch.

pattyland, 2017-03-13:
I touched `/etc/.resize_runonce` and rebooted. Partition is still 145MB, log says:

> resize2fs 1.43.3 (04-Sep-2016)
> The filesystem is already 153600 (1k) blocks long.  Nothing to do!