samtap/fang-hacks #52: Unable to select Wifi

pattyland, 2017-03-09:
Hi,

my Xiaofang is running in Access Point mode right now, and I try to connect it to my wifi. But the selection box is broken, most of the networks shows garbage information for wifis like "Address:Fon" (Fon is part of the SSID) or "Signal 7170%" (???) or "Security:WLAN" 

When I try to connect to my Wifi there is an error that the Security is not supported, I guess the information is wrong!

Greets, Sören

samtap, 2017-03-09:
Things are messed up when SSID contains a space. Could that be it?

pattyland, 2017-03-09:
Looks like that! I can ask my neighbours to change their SSID's but I'm afraid that it will work :/

samtap, 2017-03-09:
It's just the web interface that messes things up. You can manually change config files as needed. Or build a faraday cage to shield your home from foreign signals :wink:

pattyland, 2017-03-09:
While I'm searching the parts for my cage I wrote my SSID and PSK into /media/mmcblk0p2/data/etc/wpa_supplicant 
How do I activate it? Now the wifi on the network tab are complety gone :(

timwharton, 2017-04-11:
I create a PR #95 to fix this issue.

gitingear, 2017-04-27:
posted some comments related to this on issue #88