samtap/fang-hacks #54: Whats wrong with Filesystem

SatoshiNakamoto, 2017-03-10:
Hi, i dont understand, what wrong.
Based, MBR, parts are logical

First partition FAT32(Active), Second Ext2. After reboot and insert SD-card, 
http://192.168.1.79/cgi-bin/hello.cgi
System Volume Information

time_lapse

record

log_286C077AD928_s.txt

bootstrap

snx_autorun.sh

http://192.168.1.79/cgi-bin/status
404 Not Found
The requested URL /cgi-bin/status was not found on this server.

ZeroPoints, 2017-03-10:
How long did you wait before you inserted the SD after starting the cam? Did you wait till the blue light had stabilized?

SatoshiNakamoto, 2017-03-10:
yes of course... i read your man

ZeroPoints, 2017-03-10:
If you are having trouble getting it to work with the steps in the readme. Try flashing the supplied image instead of doing the manual method https://github.com/samtap/fang-hacks/releases

SatoshiNakamoto, 2017-03-10:
Thank's, replace with .img :) zip Embarrasses

closed 2017-03-11 by SatoshiNakamoto