samtap/fang-hacks #59: RTSP over HTTP

m0nit0r, 2017-03-13:
How I can get RTSP stream over HTTP?

I added "-T 8000" in snx_rtsp_server script and tried urls:(http://device-ip:8000/ and http://device-ip:8000/unicast) - It doesn't work...

What am I doing wrong?

samtap, 2017-03-13:
I've heard this from others users, apparently it simply doesn't work in the currenty rtsp server executable.

bryanjgill, 2017-03-22:
M0nit0r,

I just picked up two of the Xiaofang IP Cameras and I'm having the same problem on both of them.  What I'm really after is just connectivity to my internal wireless LAN and HTTP streaming along with setting a view -only username and password. If you happen across a solution to get HTTP streaming working, please post back here so I can give it a shot.

Thanks,

Bg