samtap/fang-hacks #60: How apply modifications?

marcoblacks, 2017-03-14:
Hi everybody! I need some help from You!!
I followed the tutorial:
- upload the image file on the SD Card;
- load the web page with hack tools;
- apply the hack (clicking on the relative easy button);
but when i tried to reboot the camera (click on the reboot button on the web page), the camera forget all the modification...
And so again:
Remove the SD card
boot the device (connecting the power plug): continuous yellow led;
Yellow led start to flashing: no wifi signal is transmitted from the camera;
Click on Setup small button (fisical on camera hole)-->voice start say some word --> default Wifi signal is ON and i can connect to the camera;
Insert the SD card (just now and not from the beginning..,)
open the hack webpage, but it tell me again to Reboot to apply the hack modification..!!
How can i apply the hack permanently and use it also after a camera switch off?
I also tried to change the wifi connection from Cloud to Wireless Client and i connected the camera to my wifi network succesfully, but again if i reboot the camera it loose all mods...
Thanks for your help!

samtap, 2017-03-14:
The hacks should be persistent, after you click 'apply' button and reboot it should work consistently. The 'apply' action copies scripts to the device and once they exist the button disappears from the status page (replaced by a message 'hacks already installed').

marcoblacks, 2017-03-14:
Many thanks for your reply!!
I will check your warnings, but i remember that after push the apply button it seems to be ok (it open another page telling that the hack is performed and i can push a 'back' button to go on the previous main page).
After that, a red writings tell me somethings like : to apply the hack a reboot is needed...

The main question is:
 -- if the hack is well performed, after reboot the camera doesn't enable any Wifi (like in the original condition), so to activate wifi the only way is push the setup button (small hole on the rear), but probably doing that, it restore also the original firmware.. removing the hack?
Thanks again!
Marco

samtap, 2017-03-14:
The setup button doesn't restore anything in the firmware so no need to worry about that. If you leave all default settings, the camera should connect using the settings entered in mi home after reboot. 
What you may be experiencing is issues related to filesystem resize? If you expand the partition you'll need to reboot to actually resize the data partition and this can take quite a while depending on your sdcard size. You can see resize2fs active in the process list and the status page will complain that data partition isn't available, until the resize2fs is completed.

marcoblacks, 2017-03-14:
Thanks a lot!! After a 75 minute waiting the resize2fs process finished!! Halleluja!!! Now all is ok, i also change from Cloud network to Client wireless mode! I don't need Xiaomi App, just a good webcam to connect to my network!
Thanks again!

marcoblacks, 2017-03-14:
Is there a way to setup a static IP on Webcam? i need to change some settings by telnet?

samtap, 2017-03-16:
I've not made any provisions for setting a static ip, so you'll have to manually edit config files.

Perhaps your router allows to setup static leases (offer the same ip on dhcp requests for a specific MAC address)?

marcoblacks, 2017-03-17:
Oh yes, i can also specify a fixed dhcp address true my router....thanks!

Tommy032017, 2017-04-03:
@samtap 
Unfortunately I can not set a new fixed IP address for a specific MAC address in the router. I only could tell the router to remember the IP address to a specific MAC address.
Would be great, if I could set my own static IP address, subnetmask, gateway and DNS by editing a config file in the camera. Does anybody know how to add a fixed IP address, or what file need to be edit?
Thanks.

samtap, 2017-04-03:
@Tommy032017 If you're up for it, you can probably make some adjustments in 01-network (remove line containing udhcpc, add ifconfig command for ip and netmask, route add default gateway, echo "1.2.3.4" > /etc/resolv.conf for DNS). Let me know how it works out so I can make a user-friendly fix for the next version!