samtap/fang-hacks #61: Camera running hot/warm

conradhagemans, 2017-03-14:
I have running the hack for more then 2 weeks now on both my 2 camera's.
I noticed that they are running warm.

Does anyone know why?

Is the sensor recording all the time even if not an RTSP stream is requested?
Isn't it possible to put the camera in idle mode till a stream is requested or motion is detected?

samtap, 2017-03-14:
My camera's don't run hot, actually they do get hot but it doesn't make any difference if I run cloud app or rtsp server. Can you monitor cpu usage (i.e. using top command in ssh), does it run hot because it uses a lot of cpu or is there another reason, i.e. the IR leds heating up the device from the inside as was suggested elsewhere?

conradhagemans, 2017-03-14:
Hi,

Thanks. I checked with top.
CPU is very low. What is nic 73.6%

Mem: 26784K used, 10172K free, 0K shrd, 4776K buff, 13936K cached
CPU:  0.3% usr 25.7% sys  0.0% nic 73.6% idle  0.0% io  0.0% irq  0.1% sirq
Load average: 0.21 0.09 0.09 1/50 2795
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 1041     1 root     S    11072 29.8  3.7 snx_rtsp_server
  570     2 root     DW       0  0.0  2.7 [isp]
 2779  2775 root     R     1356  3.6  0.4 top
  830     2 root     SW       0  0.0  0.2 [RTW_CMD_THREAD]
 1030     1 root     S     1352  3.6  0.0 ntpd -p time.google.com
 1034     1 root     S     1348  3.6  0.0 telnetd -p 2323 -l fang_login
 1039     1 root     S     1348  3.6  0.0 tcpsvd -E 0.0.0.0 21 ftpd -w /
 2795  1045 root     S     1344  3.6  0.0 sleep 3
    1     0 root     S     1164  3.1  0.0 init
  944     1 root     S     1164  3.1  0.0 udhcpc -i wlan0 -p /var/run/udhcpc.pid
 2775  2770 root     S     1160  3.1  0.0 -sh
  710     1 root     S     1160  3.1  0.0 /sbin/getty -L ttyS0 115200 vt100
 1045     1 root     S     1152  3.1  0.0 {fang-ir-control} /bin/sh /media/mmcbl
  711     1 root     S     1152  3.1  0.0 /sbin/syslogd -n -m 0
  712     1 root     S     1148  3.1  0.0 /sbin/klogd -n
  898     1 root     S     1096  2.9  0.0 wpa_supplicant -Dwext -iwlan0 -c/tmp/w
  987     1 root     S      780  2.1  0.0 /usr/boa/boa
 2770  1043 root     S      596  1.6  0.0 dropbear -R -P /var/run/dropbear.pid
 1043     1 root     S      572  1.5  0.0 dropbear -R -P /var/run/dropbear.pid
    4     2 root     SW       0  0.0  0.0 [events/0]

samtap, 2017-03-14:
%nice is related to the amount of cpu resources claimed by processes running with regular nice prio (i.e. processes that can be dropped in case something with a higher prio requires cpu, which is not really relevant for our device).

But you should read it the other way around, in "CPU: 0.3% usr 25.7% sys 0.0% nic 73.6% idle 0.0% io 0.0% irq 0.1% sirq" the 73.6% is the amount of time the cpu is idle.

MarkG55, 2017-03-15:
I think that I've worked out the cause of this issue - it's related to the IR control script - it's leaving the solenoid for the IR cut filter energised when it shouldn't be - updated fang-ir-control.sh:

```bash
#!/bin/sh

echo "IR script started"
# ir_init
gpio_ms1 -n 2 -m 1 -v 1
gpio_aud write 1 1 0
gpio_aud write 0 2 1
gpio_aud write 1 0 0

echo 0x40 > /proc/isp/filter/saturation

sleep 3

# ir loop
IR_ON=0

while :
do
        DAY="$(gpio_aud read 2)"
        if [ $DAY -eq 1 ]
        then
                if [ $IR_ON -eq 1 ]
                then
                        gpio_ms1 -n 2 -m 1 -v 1
                        gpio_aud write 1 0 0
                        echo 0x40 > /proc/isp/filter/saturation
                        IR_ON=0

                        # wait a second and then disable the output to the solenoid to reduce power use!
                        sleep 1
                        gpio_ms1 -n 2 -m 1 -v 0
                fi
        else
                if [ $IR_ON -eq 0 ]
                then
                       echo 0x0 > /proc/isp/filter/saturation
                        gpio_aud write 1 0 1
                        gpio_ms1 -n 2 -m 1 -v 0
                        IR_ON=1
                fi
        fi
        sleep 3
done
```

It's nearly midday here, so I can't actually test this until tonight, but I'm pretty confident that this is the fix.  I suspect that the reason why most people aren't seeing this issue is because they're in cooler months in the northern hemisphere - meanwhile, here in Australia, it's often over 30 degrees C at midnight!

In further testing that I've done, I've found that IR sensitivity also gets worse over time with the cloud apps - just not as bad as when running the script.  If we can find a way to further reduce power use - e.g. by undervolting/underclocking the processor then that would improve things further.

thanme, 2017-03-15:
@MarkG55 Where did you discover what the arugements for gpio_ms1 are?

TommyChausson, 2017-03-15:
@MarkG55 , good to hear that the poor vision is an issue! I've just tested your Script on one of my camera, but keep to have a sad night vision. Should we wait the camera become "colder" to view a . signifiant difference ?

MarkG55, 2017-03-15:
@thanme I'm just using the commands that are already in the IR control script - just copied one line to 2 other places.  I'm presuming that the last value indicates the digital output that will be set by the command.

@TommyChausson Yes, you'll need to turn the camera off for a while to let it cool down.  The problem is only causing extra current draw while the IR LEDs are off - once the IR LEDs are on, there is no extra current draw, but it seems that the camera just doesn't get a chance to cool down properly once it has warmed up.  I'm 100% confident that if you leave it off for 5 to 10 minutes, you'll see a significant improvement - the question is just on whether it stays that way, or whether it will slowly go bad again.

TommyChausson, 2017-03-15:
@MarkG55 , "Day is coming" so i cannot test now :( , i will try tonight

MarkG55, 2017-03-15:
OK, I've done some more testing and found that compared to the standard set up, it was actually using more power when the IR LEDs were on - I've updated the script above.  I'm pretty confident that it should be right now, but still can't check it because it isn't dark enough yet.

samtap, 2017-03-15:
Why don't you cover the lens to make it dark? ;-)

Anyway great job at debugging this. I wonder if we could also find some way to control brightness of the IR leds. I think with the current script they're either off or on, but with cloud app the brightness is controlled based on how much light is reflected. If something is close to the lens, the brightness is reduced to prevent the object from appearing too bright and white.

MarkG55, 2017-03-15:
Thanks - I just did some testing and it seems a lot better with these changes.

I had a camera running the cloud app alongside one with the updated hack and the pictures were quite similar.  The picture on the hacked version appears more grainy, but I think that just comes down to the fact that I'm running with MJPEG which I find has more detail than the H.264 stream.  The camera running the cloud app is also a newer version which has a slightly wider angle lens, so that was also making it a bit harder to compare.

Are you sure that the cloud app is controlling the brightness of the IR LEDs?  They don't appear to be varying in brightness to me and I'd expect them to simply be either on or off with either option.  To handle the case of something being close to the camera with a lot of IR being reflected, it looks like the camera in both cases simply reduces the exposure time (or maybe the gain)  When I did the test tonight, I found that the hacked version was actually giving a better picture - but I think that comes down to the difference in the angle of the view from the different lenses - sitting at a certain distance from both cameras, the hacked version was showing a good picture of me with a dark background and the unhacked version was washing me out almost completely.  When I moved closer to the cameras so that I was taking up a similar amount of the picture area on the wider angle lens version, the picture of me then came good.

Overall, even with this change, I'm still not overly impressed with the IR performance of this camera.  I'd rate it as OK for a small room, poor for a larger room and pretty much useless outside even within the rated 9m range (because you're not getting the reflection of the light that you get within a room)

For the price for these cameras though, it's still hard to complain of course...

If we can get the camera to run cooler than even with the cloud app, then I think that the night time performance would significantly improve.  If it's not possible to reduce power consumption in any way, I'm tempted to try drilling some holes in the top of one of my cameras to allow better ventilation and see how that goes!

conradhagemans, 2017-03-16:
I am pleased with the results. I updated both my camera's with the modified script.
It's night here. So the first picture is without lights on in the room, the second with the lights on.

[image]
[image]

thanme, 2017-03-16:
So I was playing with this last night (I'm using your original update and not the new one yet) and it looks great.
I'm currently running an unhacked camera next to a hacked one to compare and I put my hand over them to check for heat. By doing so, I covered the lens which caused the IR to turn off on the unhacked version and nothing on hacked on. having something that close to the lens with IR on made it all white which triggered something.
Does anyone have any idea how I can get a console output for the cloud apps?? I'm keen to see if we can work out what's actually happening

SuperSaiyan1, 2017-03-17:
@conradhagemans 
Hi Conrad.  Can you share your config of homeassistant and xiaofang camera. Having a lot of trouble with it.

conradhagemans, 2017-03-17:
Yes ofcourse I can share my setup.
There are two ways to integrate your camera's within HomeAssistant

Method 1: Direct connection. Probably the easiest one. 
In your configuration.yaml put the following lines:

```
camera:
  - platform: ffmpeg
    input: rtsp://192.168.1.205/unicast
    name: CAM Mini
  - platform: ffmpeg
    input: rtsp://192.168.1.191/unicast
    name: CAM Mini 2
```
Ofcourse put your own IP addresses in.

Method 2: Indirect via Zoneminder. I like this method more. Then you are also able to trigger recording.

In configuration.yaml:

```
zoneminder:
  host: localhost
  username: zoneminder_user
  password: zoneminder_user_password

camera:
   - platform: zoneminder

switch:
  - platform: zoneminder
    command_on: Record
    command_off: Nodect
```
Do you know how to setup Zoneminder configuration to see our camera's?

conradhagemans, 2017-03-17:
If you want a more fancy setup in Homeassistant like on my images to have camera button on the top menu etc. let me know then I share my full setup/config files.

samtap, 2017-03-17:
@conradhagemans this stuff can be really useful to others, would you mind putting it up on the wiki? This issue will likely be closed some day making it much harder to find.

I will make sure the fixes for the ir control script will be included in the next image.

nikonaum, 2017-03-18:
@conradhagemans ZoneMinder setup: 
Set the source type to FFmpeg
Source Path: rtsp://[your-ip]/unicast
Remote Method RTP/RTSP

Tommy032017, 2017-03-18:
First thanks to samtap for the great work and investigations. I use 5 XiaoFang.
I also have a dark bad stream in night view over rtsp after a period of time. But I the recorded videos on sd-card are normal and looks like the original quality without the hack. Really strange behavior. 
What I did:
I programmed the camera with motion detection in MiHome before installing the hack. Then the camera will use the boot partition to add 2 folders records and timelaps. In records will be stored the motion detected videos. The stored videos are pretty fine, while the stream is awful dark, blurry and grainy over rtsp in vlc.
I don't know if this will be something within temperature, ffmpeg or CPU usage.   
@conradhagemans 
I tried your changes in the IR-script, but did not get the script to run. I always get status NOK by starting the script. I copied the new lines to /data/usr/bin/fang-ir-control.sh, but the script gives me always a NOK.
I tried this changes by ftp and also by editing the file in the image. Any suggestions what I did wrong.
Thanks.

nehaleem, 2017-03-19:
Hi, could be possible to merge this fix to the new version ? I tried version 0.2 but i'ts really overheating (i can't hold hand on front glass) and camera will freeze because of that heat.

ZeroPoints, 2017-03-19:
@nehaleem Did you even read the thread. Samtap said he will make sure to include them in next.

iothing, 2017-03-20:
I submitted a pull request that fixes this issue, if you cannot wait: https://github.com/iothing/fang-hacks

nehaleem, 2017-03-20:
@ZeroPoints Sorry, didn't saw that. Don't get me wrong, i'm really grateful for all this work! I just asked :-)

Tommy032017, 2017-03-20:
@iothing & @samtap 
Thanks for your efforts and the new script to fix the overheating, but now I will have a purple night view and no more b/w. I did some efforts and may be found the problem.
It seams to be something to the "gpio_ms1 -n 2 -m 1 -v 0" when switching IR on.

I did this changes and hopefully it will work. At the moment temperature is normal and night view normal b/w.

```ruby

<#!/bin/sh

echo "IR script started"
# ir_init
#gpio_ms1 -n 2 -m 1 -v 1 # this causes increased current flow
gpio_ms1 -n 2 -m 1 -v 0 # has something to do with the ir-cut/pass filter movement
gpio_aud write 1 1 0    # pin 1 is an output and is set to low, purpose unknown
gpio_aud write 0 2 1    # pin 2 is an input, the photoresistor
gpio_aud write 1 0 0    # pin 0 is an output and set to low, these are the ir-leds

sleep 3

# ir loop
IR_ON=0

while :
do
        DAY="$(gpio_aud read 2)"
        if [ $DAY -eq 1 ]
        then
                if [ $IR_ON -eq 1 ]
                then
                        gpio_ms1 -n 2 -m 1 -v 1 # filter movement enabled
                        gpio_aud write 1 0 0    # disable ir led and latch the filter in the correct position
                        #gpio_ms1 -n 2 -m 1 -v 0 # filter movement disabled
                        echo 0x40 > /proc/isp/filter/saturation
                        IR_ON=0
                fi
        else
                if [ $IR_ON -eq 0 ]
                then
                       echo 0x0 > /proc/isp/filter/saturation
                        #gpio_ms1 -n 2 -m 1 -v 0 # filter movement enabled
                        gpio_aud write 1 0 1    # enable ir led and latch the filter in the correct position
                        gpio_ms1 -n 2 -m 1 -v 0 # filter movement disabled
                        #echo 0x40 > /proc/isp/filter/saturation
                        IR_ON=1
                fi
        fi
        gpio_ms1 -n 2 -m 1 -v 0 # filter movement disabled
        sleep 3
done
```

Tommy032017, 2017-03-20:
I think the color problem comes from "echo 0x40 > /proc/isp/filter/saturation":
Could it be, that:
echo 0x0 > /proc/isp/filter/saturation  = b/w mode
echo 0x40 > /proc/isp/filter/saturation  = color mode

Tommy032017, 2017-03-20:
Unfortunately, I've been glad too early - temperature on lens is after 2h again by 50.5 °C.

SuperSaiyan1, 2017-03-21:
@conradhagemans 

I would appreciate if you cloud share your config.

I have a problem viewing the stream in Homeassistant (RP3)
The preview pic changes every 10 - 15 seconds and  that's OK.
[image]

But when I open the strem and watch the video. It goes blank.
[image]

Tried with your setup:
`camera 1:
  - platform: ffmpeg
    input: rtsp://192.168.1.142/unicast
    name: CAM Mini`

And also 
`camera 1:
  - platform: ffmpeg
    input: -rtsp_transport tcp -i rtsp://192.168.1.142/unicast
    extra_arguments: -pred 1 -q:v 2
    name: CAM Mini`

conradhagemans, 2017-03-21:
@SuperSaiyan1 No problem.
My camera's do not run very warm. I made some changes to the 20-rtsp-server script on the camera. With the default script VLC was crashing. Maybe the same happens with Homeassistant.

The script is stored in this directory: /media/mmcblk0p2/data/etc/scripts

```
#!/bin/sh
PIDFILE="/var/run/rtsp-server.pid"

status()
{
  pid="$(cat "$PIDFILE" 2>/dev/null)"
  if [ "$pid" ]; then
    kill -0 "$pid" >/dev/null && echo "PID: $pid" || return 1
  fi
}

start()
{
  LOG=/dev/null
  echo "Starting RTSP server..."
#  snx_rtsp_server -W 1920 -H 1080 -Q 10 -b 4096 -a >$LOG 2>&1 &
  snx_rtsp_server >$LOG 2>&1 &
#  snx_rtsp_server -W 640 -H 480 -j 60 -Q 5 -u media/stream1 -P 8554 >$LOG 2>&1 &
/media/mmcblk0p2/data/test/snx_isp_ctl --osdset-en 1 --osdset-ts 1 --osdset-template 1234567890./-:Date --osdset-gain 2 --osdset-bgtransp 0x1 --osdset-bgcolor 0x000000
  echo "$!" > "$PIDFILE"
}

stop()
{
  pid="$(cat "$PIDFILE" 2>/dev/null)"
  if [ "$pid" ]; then
     kill $pid ||  rm "$PIDFILE"
  fi
}

if [ $# -eq 0 ]; then
  start
else
  case $1 in start|stop|status)
    $1
    ;;
  esac
fi
```
As you can see I am running the snx-rtsp-server without any parameters. I think it will run then in 720p mode and not 1080p. Then in Homeassistant with this setup:

```
camera:
  - platform: ffmpeg
    input: rtsp://192.168.1.205/unicast
    name: CAM Mini
```

the stream will run smooth. Maybe therefore the camera isn't running hot.
Did you test to connect with VLC?

To test in VLC choose in the menu 'Open Network' then fill in rtsp://192.168.1.205/unicast

[image]

Good luck. Let us know if you get it working.

Tommy032017, 2017-03-21:
@conradhagemans 
Would be kind, if you could tell me, what script you are using for fang-ir-control.sh? 
Yesterday I tried a lot of different settings within the script, but the camera lens heat up over 50 °C (room temp. about 21 °C).
May be you could provide me a complete image (or only the changed files of fanghacks_v0.2.0) of your using hack. 

Thank you very much.

iothing, 2017-03-21:
Tommy032017: apologies, this was my fault with the purple image, I submitted a new pull request fixing this

For your camera running hot: could you try to measure the current? Also check top to see if there is some process causing heavy cpu load that could result in heat.

conradhagemans, 2017-03-21:
@Tommy032017 and @Yannik25 The only change I made was in the 20-rtsp-server script. I am using default image from fang-hacks. My ir-control script:

```
Contents of script '21-ir-control':

#!/bin/sh
PIDFILE="/var/run/ir-control.pid"

status()
{
  pid="$(cat "$PIDFILE" 2>/dev/null)"
  if [ "$pid" ]; then
    kill -0 "$pid" >/dev/null && echo "PID: $pid" || return 1
  fi
}

start()
{
  echo "Starting IR script..."
  fang-ir-control.sh /dev/null 2>&1 &
  echo "$!" > "$PIDFILE"
}

stop()
{
  pid="$(cat "$PIDFILE" 2>/dev/null)"
  if [ "$pid" ]; then
     kill $pid || rm "$PIDFILE"
  fi
}

if [ $# -eq 0 ]; then
  start
else
  case $1 in start|stop|status)
    $1
    ;;
  esac
fi
```
My result from top:

```
Mem: 26808K used, 10148K free, 0K shrd, 4608K buff, 13904K cached
CPU:  0.7% usr  2.9% sys  0.0% nic 95.8% idle  0.0% io  0.0% irq  0.3% sirq
Load average: 0.01 0.05 0.05 1/52 12465
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 1077     1 root     S    11044 29.7  1.8 snx_rtsp_server
  570     2 root     SW       0  0.0  0.9 [isp]
12410 12401 root     R     1356  3.6  0.5 top
 1029     1 root     S     1172  3.1  0.3 {miio_client_hel} /bin/sh /etc/miio_cl
12380  1079 root     S      596  1.6  0.1 dropbear -R -P /var/run/dropbear.pid
  836     2 root     SW       0  0.0  0.1 [RTW_CMD_THREAD]
 1065     1 root     S     1352  3.6  0.0 ntpd -p time.google.com
 1075     1 root     S     1348  3.6  0.0 tcpsvd -E 0.0.0.0 21 ftpd -w /
 1069     1 root     S     1348  3.6  0.0 telnetd -p 2323 -l fang_login
12465  1081 root     S     1344  3.6  0.0 sleep 3
    1     0 root     S     1164  3.1  0.0 init
  942     1 root     S     1164  3.1  0.0 udhcpc -i wlan0 -p /var/run/udhcpc.pid
12401 12380 root     S     1160  3.1  0.0 -sh
  710     1 root     S     1160  3.1  0.0 /sbin/getty -L ttyS0 115200 vt100
 1081     1 root     S     1152  3.1  0.0 {fang-ir-control} /bin/sh /media/mmcbl
  711     1 root     S     1152  3.1  0.0 /sbin/syslogd -n -m 0
  712     1 root     S     1148  3.1  0.0 /sbin/klogd -n
12463  1029 root     S     1144  3.0  0.0 sleep 1
  897     1 root     S     1100  2.9  0.0 wpa_supplicant -Dwext -iwlan0 -c/tmp/w
  995     1 root     S      780  2.1  0.0 /usr/boa/boa
 1079     1 root     S      572  1.5  0.0 dropbear -R -P /var/run/dropbear.pid
  409     2 root     SW       0  0.0  0.0 [mtdblock2]
    4     2 root     SW       0  0.0  0.0 [events/0]
  518     2 root     SWN      0  0.0  0.0 [jffs2_gcd_mtd4]
  211     2 root     SW       0  0.0  0.0 [bdi-default]
  209     2 root     SW       0  0.0  0.0 [sync_supers]
  593     2 root     SW       0  0.0  0.0 [mmcqd]
  223     2 root     SW       0  0.0  0.0 [khubd]
    5     2 root     SW       0  0.0  0.0 [khelper]
  270     2 root     SW       0  0.0  0.0 [kswapd0]
  269     2 root     SW       0  0.0  0.0 [khungtaskd]
  231     2 root     SW       0  0.0  0.0 [kmmcd]
  241     2 root     SW       0  0.0  0.0 [cfg80211]
  316     2 root     SW       0  0.0  0.0 [aio/0]
  262     2 root     SW       0  0.0  0.0 [rpciod/0]
  414     2 root     SW       0  0.0  0.0 [mtdblock3]
```
I noticed that there is still a process running "miio_client_hel" I am not shure what that is.
I can try to measure the current with an usb monitor.
First image is with ligths on, second with lights off.

[image]

[image]

Tommy032017, 2017-03-22:
@iothing: Thank you. No problem at all. I now use your latest script. Will see what happened next days.
@conradhagemans: Thank you. I do not have an usb meter and do not want to cut a cable to use my normal power meter. So I could not do any measurements of current.
I also did a capture with latest Fang Hacks Version 0.2.0 and Wireshark only show an external connection to NTP server. So it seams everything will be ok and privacy.

My top is: (but do not see mem, cpu .... Where to find this information?)
```
 PID USER       VSZ STAT COMMAND
    1 root      1164 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [events/0]
    5 root         0 SW   [khelper]
    8 root         0 SW   [async/mgr]
  209 root         0 SW   [sync_supers]
  211 root         0 SW   [bdi-default]
  213 root         0 SW   [kblockd/0]
  223 root         0 SW   [khubd]
  226 root         0 SW   [kseriod]
  231 root         0 SW   [kmmcd]
  241 root         0 SW   [cfg80211]
  262 root         0 SW   [rpciod/0]
  269 root         0 SW   [khungtaskd]
  270 root         0 SW   [kswapd0]
  316 root         0 SW   [aio/0]
  323 root         0 SW   [nfsiod]
  332 root         0 SW   [crypto/0]
  399 root         0 SW   [mtdblock0]
  404 root         0 SW   [mtdblock1]
  409 root         0 SW   [mtdblock2]
  414 root         0 SW   [mtdblock3]
  419 root         0 SW   [mtdblock4]
  424 root         0 SW   [mtdblock5]
  429 root         0 SW   [snx-spi.0]
  438 root         0 SW   [zd1211rw]
  485 root         0 SW   [usbhid_resumer]
  518 root         0 SWN  [jffs2_gcd_mtd4]
  567 root         0 SW   [isp]
  593 root         0 SW   [mmcqd]
  647 root       816 S    /tmp/boa
  734 root      1160 S    /sbin/getty -L ttyS0 115200 vt100
  735 root      1152 S    /sbin/syslogd -n -m 0
  736 root      1148 S    /sbin/klogd -n
  737 root         0 SW   [RTW_CMD_THREAD]
  740 root      1088 S    wpa_supplicant -B -i wlan0 -c /media/mmcblk0p2/data/
  748 root      1364 S    udhcpc -i wlan0 -p /var/run/udhcpc.pid -b
  751 root      1352 S    ntpd -p time.google.com
  755 root      1348 S    telnetd -p 2323 -l fang_login
  760 root      1348 S    tcpsvd -E 0.0.0.0 21 ftpd -w /
  762 root     10976 S    snx_rtsp_server
  764 root       572 S    dropbear -R -P /var/run/dropbear.pid
  765 root      1152 S    {fang-ir-control} /bin/sh /media/mmcblk0p2/data/usr/
 1183 root      1344 S    sleep 3
 1184 root      1164 S    {status} /bin/sh /tmp/www/cgi-bin/status
 1190 root      1164 S    {status} /bin/sh /tmp/www/cgi-bin/status
 1191 root      1144 S    cat
 1200 root      1152 R    ps
```

I have 5 cameras, ordered last month over gearbest, called iSC5 from mi.com, commit ID: 2016DP5557.
On 3 packages is are labels QC49 and US. The others are Chinese versions.

SuperSaiyan1, 2017-03-22:
@conradhagemans 

After changing the **20-rtsp-server script** the camera works in HA. Thank you very much 👍

The Zoneminder looks interesting. I will try to set it up in HA. Do you have any pointers.

ZeroPoints, 2017-03-22:
Don't know what that ^ all had to do with Hot/Warm running. I have been using the script in this chat history and it still runs hot i've opened mine up and the board with the sonix chip is on(that takes the power input) is the board running hot. The camera and its board and and board with the IR are fine.

Tommy032017, 2017-03-22:
Looking for the overheating problem, I opened my camera today. The PCB's are from 02-2017! The Sonix heats up to 65°C in normal use with the hacks.  Thanks for the hint, @ZeroPoints 
At the moment I have 3 considerations:
1. Adding a cooling heatsink to the Sonix chip, maybe the same like for raspberry.
2. The camera only supports 720P.
3. There must be different running processes, which are responsible for the overheating, combined to the original firmware. I disabled all selectable scripts, but without any difference in temperature.

MarkG55, 2017-03-23:
@Tommy032017 I've been doing some further testing on this.  

Even with my modified IR control script, the camera does still heat up quite a lot - just not to the same extent as it used to.

I had planned to try attaching a heatsink to the Sonix chip to see what difference that made, but when I went to buy one the shop was out of stock!  I'll have to try again.

In the mean time, I've got the following observations:

* Current draw with the hack is the same as with the hack disabled.  I haven't got a camera that has never been modified to measure the current of a camera in its completely original state - I've got several more cameras on the way, so I'll do that test once they arrive.
* Even with the hack disabled, the camera still does get very warm and the night picture still degrades even with the original cloud app running.  Picture quality between a camera with the hack disabled and one hacked with the modified IR script seems pretty much the same as far as I can tell.
* I did a test with the camera "naked" - i.e. I pulled the guts out and had it sitting outside the plastic shell - I just had the plastic shell sitting beside it so that the antenna could be left attached - this resulted in the Sonix chip running much cooler and allowed for a **much** brighter night image.  Interestingly, the image appeared more blurry - it would seem that thermal noise is actually helping improve the effective resolution of the image which is rather weird.
* Reassembling the camera but leaving the back piece of plastic off to allow for a bit more air flow showed an improvement over having it in place, but nowhere near as much as having the entire enclosure taken off.
* The holes at the top of the back plastic piece are actually blocked by the little speaker - I tried removing the speaker to see how much difference that would make and I was suprised to find that there was no noticeable difference compared to having the speaker there.

I'm hopeful that adding a small heatsink to the Sonix chip will solve the problem.  If it doesn't solve the problem completely, I'll then try drilling some holes in the top of the plastic enclosure to see if that helps any.

ZeroPoints, 2017-03-23:
I think it needs some direct air flow flowing through the device  or a heatsink in its current state. Doesnt look like much room between chip and back piece though in enclosure

conradhagemans, 2017-03-23:
I would like to come back on the heat issues. I am reading that some camera's getting as hot as 70 C. What I don't understand why mine aren't that hot. I have two camera's.
When I pick them up by hand I do feel that they are internal a bit warm.
Do I have an other build or version?
Or is it a design flaw and are the camera's always to warm even running in non-hacked mode?

conradhagemans, 2017-03-23:
One more question.
I am running the snx_rtsp_server without any parameters in the 20-rtsp-server script. I noticed that the ouput stream is then only 720p. Could that be the reason my camera's aren't running hot?

Tommy032017, 2017-03-23:
I also did some tests yesterday and even without running the selectable scripts the camera will get temperature at the lens outside (with case closed) of about 50-55°C and the Sonix goes up to 65-70°C (back cover open) . With the original firmware the lens has about 35°C and the Sonix about 45-50°C. I also tried the rtsp script with no options (witch standard settings it will run 720p), but temperatures are the same within the 1080p options. So no difference by changing from 1080p to 720p.
But even without running any script of the hack (selectable ones), the Sonix heat up to 65-70°C.
Doing some cooling with heatsink or drilling holes will not solve our main problem. There must be anything of programming or scripting, which trigger the high temperatures of the Sonix. Our goal should be to find the processes which are responsible for this.
May be in the current hack is something overtaken from a other hacks, which do some overclocking to the chip?
May be the last solution could be to use the factory default P2P stream, and only change networking from access point to client mode. Pls. see: https://github.com/samtap/fang-hacks/issues/71
I would work on this solutions, but I do not have programming experience. So may be samtap and some others could help us.

MarkG55, 2017-03-23:
@Tommy032017 With the camera that you have got that isn't running hot, has it ever had the hack enabled on it?

You could well be right that the hack is causing something to change and that's causing it to overheat, and disabling the hack through the web interface isn't setting it back sufficiently to how it originally was.

samtap, 2017-03-23:
I haven't really kept track of all issues related to overheating / ir script. There was a pull request a while back that was merged but apparently it's not fixed yet? I'll do some testing myself this weekend since I haven't been able to reproduce it thus far. I have a hard time believing the camera heats up while no scripts and/or cloud apps are running. I also wouldn't be surprised if it only occurs in some camera, i.e. those with a different hardware revision or firmware version.

Temperatures above ~50 degrees C. can be considered irregular, if your camera runs this hot please capture and share some info so we can finally resolve this:
- Describe your situation, i.e. which firmware version, ir-control script, fang-hacks settings and possible other customizations.
- Save output of 'uptime' and 'ps' commands

Tommy032017, 2017-03-23:
@MarkG55: Yes, the camera runs normal, after going back to original firmware. I installed the hack and temperature goes up and then I restarted the camera with fang_hacks_rescue.cfg. After waiting a minute, I removed power and sd card and pluged in usb power again while pressing the setup button for about 20 seconds until voice comes to read QR. After that, I did normal connection to MiHome and camera works again with normal temperatures.
I used the normal hack.
fanghacks_v0.2.0.img  = Sonix went to ~70°C
fanghacks_v0.2.0.img with different modified fang-ir-control.sh = Sonix went to ~70°C
going back to stock = Sonix went to ~ 45°C

At the moment I have running the camera with stock firmware as access point (select ismart camera... in MiHome) and there are no abnormal temperatures and brilliant pictures.

Tommy032017, 2017-03-23:
@samtap: Thank you very much to do some more investigations to find a solution to the heating problems.
I'll give you of course all my collected information (also added some pictures to the PCB's)
MiHome = Android V 4.1.7 with Plugin version 2.6.4
Camera Firmware = 3.0.3.56
Using stock firmware, temperatures outside the housing are back=41°C, top=45°C, lens=42°C, room temp. = 22°C
Overheating:
fanghacks_v0.2.0.img = Sonix goes to ~70°C
fanghacks_v0.2.0.img with different modified fang-ir-control.sh = Sonix goes to ~70°C
going back to stock = Sonix goes to ~ 45°C

> Describe your situation, i.e. which firmware version, ir-control script, fang-hacks settings and possible other customizations. Describe rtsp server usage as well (how many clients were connected, does temperature decrease by connecting fewer or no clients?)
Save output of 'uptime' and 'ps' commands
>
First, I installed the new camera to MiHome and did the Update to 3.0.3.56. Camera runs about 1 day without any problems. Then I installed your fanghacks_v0.2.0.img on a SanDisk 16Gb, Class4.
I did first activate the hack, then changed network to client and last switched cloud off.
A couple ours later, I looked at the b/w night stream and video was pretty darken (as there are no leds on) and the video was blurry and most grainy on the left side of the picture, also grainy on the right edge.
I did some more tests, which did not change anything to the heating problems. First, I thought that it will come from the ir cut filter, but I tested several different settings in the script without any reducing the temperature.
Heating is independent of: (at my experience)

- fang-ir-control.sh 
- switching on or of scripts in the boa
- day or night view (may be b/w view heats up some degrees, but it is worth)
- streaming over VLC or not, does not cooling down the camera. 
- only switch off rtsp script does not change anything

```
Mem: 19836K used, 17120K free, 0K shrd, 3784K buff, 7696K cached
CPU:  0.7% usr  4.3% sys  0.0% nic 94.4% idle  0.0% io  0.0% irq  0.3% sirq
Load average: 0.08 0.10 0.13 1/50 4613
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
  762     1 root     S    20976 56.5  4.2 snx_rtsp_server -W 1920 -H 1080 -Q 10
  570     2 root     SW       0  0.0  1.0 [isp]
 4597  4594 root     R     1356  3.6  0.2 top
  737     2 root     SW       0  0.0  0.0 [RTW_CMD_THREAD]
  748     1 root     S     1364  3.6  0.0 udhcpc -i wlan0 -p /var/run/udhcpc.pid
  751     1 root     S     1352  3.6  0.0 ntpd -p time.google.com
  760     1 root     S     1348  3.6  0.0 tcpsvd -E 0.0.0.0 21 ftpd -w /
  755     1 root     S     1348  3.6  0.0 telnetd -p 2323 -l fang_login
 4613   765 root     S     1344  3.6  0.0 sleep 3
    1     0 root     S     1164  3.1  0.0 init
 4594  4517 root     S     1160  3.1  0.0 -sh
  734     1 root     S     1160  3.1  0.0 /sbin/getty -L ttyS0 115200 vt100
  765     1 root     S     1152  3.1  0.0 {fang-ir-control} /bin/sh /media/mmcbl
  735     1 root     S     1152  3.1  0.0 /sbin/syslogd -n -m 0
  736     1 root     S     1148  3.1  0.0 /sbin/klogd -n
  740     1 root     S     1088  2.9  0.0 wpa_supplicant -B -i wlan0 -c /media/m
  645     1 root     S      796  2.1  0.0 /tmp/boa
 4517   763 root     S      596  1.6  0.0 dropbear -R -P /var/run/dropbear.pid
  763     1 root     S      572  1.5  0.0 dropbear -R -P /var/run/dropbear.pid
  518     2 root     SWN      0  0.0  0.0 [jffs2_gcd_mtd4]
```
Please tell me, if you need more information. Thank you again.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

samtap, 2017-03-23:
Thanks! Did you also test with the ir script completely disabled/removed? (please confirm it's not running by checking process list).
Not sure if you have background in Linux, so I'll explain: 'uptime' shows the average load of the device during the last 1, 5 and 15 minutes. Anything above 1.0 means the device could not keep up with the demand, which in most cases means the CPU is busy with 100% load, producing heat in the sonix chip.

Tommy032017, 2017-03-23:
Thanks for explanations. I uses linux mint since years, but not involved in programming.
I just started the camera with the hack again and did a stop and then a disable to the ir-control and the reboot the camera. Now it shows no PID and fields are blank. I tested the cameras light sensor and the camera do not move the ir cut filter, nor switches to b/w or turns on ir leds.
I will now wait some time and see what happened to the Sonix and will do top in ssh. 
Are this the information you meant?

Tommy032017, 2017-03-23:
Now I started the camera without rtsp and ir-control scripts. (every time rebooting or starting the camera, I'll hear  a click. May be the IR cut filter? But ir-control script is off and no switching to b/w by low lights.)
The camera runs about 10 minutes with this results:
Sonix is again by ~ 64°C (always measured with a calibrated IR thermometer, and back cover of camera opened) 
[image]
```
Mem: 16332K used, 20624K free, 0K shrd, 3092K buff, 5992K cached
CPU:  0.0% usr  0.3% sys  0.0% nic 99.6% idle  0.0% io  0.0% irq  0.0% sirq
Load average: 0.00 0.01 0.01 1/45 765
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
  765   764 root     R     1356  3.6  0.2 top
  737     2 root     SW       0  0.0  0.2 [RTW_CMD_THREAD]
  748     1 root     S     1364  3.6  0.0 udhcpc -i wlan0 -p /var/run/udhcpc.pid
  751     1 root     S     1352  3.6  0.0 ntpd -p time.google.com
  755     1 root     S     1348  3.6  0.0 telnetd -p 2323 -l fang_login
  760     1 root     S     1348  3.6  0.0 tcpsvd -E 0.0.0.0 21 ftpd -w /
    1     0 root     S     1164  3.1  0.0 init
  764   763 root     S     1160  3.1  0.0 -sh
  734     1 root     S     1160  3.1  0.0 /sbin/getty -L ttyS0 115200 vt100
  735     1 root     S     1152  3.1  0.0 /sbin/syslogd -n -m 0
  736     1 root     S     1148  3.1  0.0 /sbin/klogd -n
  740     1 root     S     1088  2.9  0.0 wpa_supplicant -B -i wlan0 -c /media/m
  648     1 root     S      780  2.1  0.0 /tmp/boa
  763   762 root     S      596  1.6  0.0 dropbear -R -P /var/run/dropbear.pid
  762     1 root     S      572  1.5  0.0 dropbear -R -P /var/run/dropbear.pid
  409     2 root     SW       0  0.0  0.0 [mtdblock2]
  593     2 root     SW       0  0.0  0.0 [mmcqd]
  518     2 root     SWN      0  0.0  0.0 [jffs2_gcd_mtd4]
    5     2 root     SW       0  0.0  0.0 [khelper]
    4     2 root     SW       0  0.0  0.0 [events/0]
```

MarkG55, 2017-03-24:
@Tommy032017 Thanks for that - I've just gone through the same process to reset everything back to how it originally was and it hasn't made any difference for me - the camera is still drawing the same amount of current (about 350 mA) and the Sonix chip is still getting too hot to hold my finger on, so it's well above 50 C for me.

Since @Tommy032017 doesn't have a USB current meter, it would be interesting if someone else who has a camera that isn't running hot could run a test to see what current it should be drawing.  I've got more cameras on the way but since they're on the slow boat from China, they could be a week or 2 at least before they get here.

For the camera set back to the original settings but still running hot:

```
Mem: 35292K used, 1664K free, 0K shrd, 2716K buff, 15524K cached
CPU:  6.6% usr 26.6% sys  0.0% nic 20.0% idle 46.6% io  0.0% irq  0.0% sirq
Load average: 0.08 0.15 0.19 1/83 2611
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
  628   619 root     S    64792174.7 24.9 /etc/app/iCamera
 2608  2598 root     R     1156  3.1  8.3 top
  619     1 root     S    10772 29.0  0.0 /etc/app/iCamera
  630     1 root     S     4020 10.8  0.0 /etc/app/test_UP
  818     1 root     S     2152  5.8  0.0 /etc/miio_client/miio_client -D
  819     1 root     S     1172  3.1  0.0 {miio_client_hel} /bin/sh /etc/miio_client/miio_client_helper_nomqtt.sh
    1     0 root     S     1164  3.1  0.0 init
  773     1 root     S     1164  3.1  0.0 udhcpc -i wlan0 -p /var/run/udhcpc.pid -b
 1344     1 root     S     1160  3.1  0.0 /usr/sbin/telnetd
 2598  1344 root     S     1160  3.1  0.0 -sh
  611     1 root     S     1160  3.1  0.0 /sbin/getty -L ttyS0 115200 vt100
  612     1 root     S     1152  3.1  0.0 /sbin/syslogd -n -m 0
  613     1 root     S     1148  3.1  0.0 /sbin/klogd -n
  748     1 root     S     1092  2.9  0.0 wpa_supplicant -Dwext -iwlan0 -c/tmp/wpa_supplicant.conf -B
  799     1 root     S      780  2.1  0.0 /usr/boa/boa
  918   819 root     S      676  1.8  0.0 /etc/miio_client/miio_recv_line
  567     2 root     DW       0  0.0  0.0 [isp]
  707     2 root     SW       0  0.0  0.0 [RTW_CMD_THREAD]
  409     2 root     SW       0  0.0  0.0 [mtdblock2]
 1154     2 root     SW       0  0.0  0.0 [mmcqd]
  223     2 root     SW       0  0.0  0.0 [khubd]
  518     2 root     SWN      0  0.0  0.0 [jffs2_gcd_mtd4]
  270     2 root     SW       0  0.0  0.0 [kswapd0]
 1219     2 root     SW       0  0.0  0.0 [flush-179:0]
    5     2 root     SW       0  0.0  0.0 [khelper]
  209     2 root     SW       0  0.0  0.0 [sync_supers]
  231     2 root     SW       0  0.0  0.0 [kmmcd]
    4     2 root     SW       0  0.0  0.0 [events/0]
  211     2 root     SW       0  0.0  0.0 [bdi-default]
  241     2 root     SW       0  0.0  0.0 [cfg80211]
  316     2 root     SW       0  0.0  0.0 [aio/0]
  226     2 root     SW       0  0.0  0.0 [kseriod]
  399     2 root     SW       0  0.0  0.0 [mtdblock0]
  404     2 root     SW       0  0.0  0.0 [mtdblock1]
  262     2 root     SW       0  0.0  0.0 [rpciod/0]
  269     2 root     SW       0  0.0  0.0 [khungtaskd]
  419     2 root     SW       0  0.0  0.0 [mtdblock4]
  424     2 root     SW       0  0.0  0.0 [mtdblock5]
  429     2 root     SW       0  0.0  0.0 [snx-spi.0]
  414     2 root     SW       0  0.0  0.0 [mtdblock3]
  485     2 root     SW       0  0.0  0.0 [usbhid_resumer]
    2     0 root     SW       0  0.0  0.0 [kthreadd]
    3     2 root     SW       0  0.0  0.0 [ksoftirqd/0]
  438     2 root     SW       0  0.0  0.0 [zd1211rw]
  332     2 root     SW       0  0.0  0.0 [crypto/0]
    8     2 root     SW       0  0.0  0.0 [async/mgr]
 2459     2 root     SW       0  0.0  0.0 [flush-mtd-unmap]
  323     2 root     SW       0  0.0  0.0 [nfsiod]
  213     2 root     SW       0  0.0  0.0 [kblockd/0]
```

Uptime:
```
 09:37:54 up 19 min,  1 users,  load average: 0.14, 0.16, 0.19
```

Tommy032017, 2017-03-24:
@MarkG55 Thanks for your testings. I did pretty the same test yesterday evening with the same result.
I tried again to recover to the stock settings of the camera. Switching on cloud, disable hack and then power off the camera, remove SD card and powered it on by pressing the setup button until voice to scan QR comes back. Then I tried to connect it again with MiHome. I tried several times to get the camera connected back to Xiaomi cloud, but it worked random once a time. Now I could not get any more connection to the cloud (blue led blinking), but the camera is connected to the local network. The camera starts with orange led then switch to blinking orange/blue and then stays only blue blinking, which means connected local, but not connected to the cloud. This start procedure seams to be normal and is the same as unhacked cameras will do. 

- Blinking orange/blue = try to connect with local network
- blue blinking = connected to local network, but has no established connection with the cloud.
- blue = connected to the cloud

Setup the camera as an access point to MiHome works also fine again. 
**But the camera runs again hot!**

So I  think we have the same problems, like discussed by the Yi-Cam.
There must be some changes by the hack which won't be reversible!

Main questions and may be the solution to the heating problems will be to find the changes which are be done to the original firmware by the hack.
1. What does the hack change, and will not be reversible? Any of this changes will be responsible for the heating problems!
2. How to come back to the factory settings? 
3. Does the camera runs at normal temperatures again after recovery? If not, the hack has damaged hardware.

Tommy032017, 2017-03-24:
I think I found a solution to go back to stock configuration without the hacks.
It could be done by snx_autorun.sh (thanks to samtap for the hint).
The script could do some changes in the camera:

First 3 files must be deleted:
TZ in /etc
TZ in /root/etc_default
rcS in /root/etc_default/init.d
Then the 3 original files must be renamed:
TZ~ renamed to TZ in /etc 
TZ~ renamed to TZ in /root/etc_default
rcS~ renamed to rcS in /root/etc_default/init.d

Then the complete content in /etc have to be deleted.
And last all files from /root/etc_default must be copied (with -a) to /etc
Then removing the SD card and power off and again on should bring back the camera to its original behavior.
May be somebody could write such a script as snx_autorun.sh for rescue.
I would do this, but I have no experience in writing scripts.

MarkG55, 2017-03-25:
@Tommy032017 As far as I can see, pressing and holding the reset button during power on does actually reset everything in /etc back to the defaults.

With regard to your suggested instructions, the first few steps are either unnecessary or won't actually work:  There's no point in deleting and renaming the TZ files at the start because your instructions end up deleting the recopying files over the top at the end anyway.  It's also not possible to rename/delete files in /root/etc_default because that's on a read only file system. 

You may well be right that the hardware has been damaged and that's why it's still drawing excess current even after resetting everything back to the defaults.  Hopefully it won't be long until I get my hands on some new cameras to do further testing...

samtap, 2017-03-25:
I did some testing today, placed my two cameras in a cardboard box to block any light and ran one with original cloud apps and one with rtsp server with two clients permanently connected. After 24h both camera's are about the same temperature, warm but not hot, definitely not as hot as reported here. I'm also using the original ir control script (as released in 0.2.0 image). So I'm out of ideas. When I get the change I'll take one of them apart and make some close-up pics to compare to ones posted by @Tommy032017

Tommy032017, 2017-03-25:
@MarkG55 Thanks. 
Trying to find a solution I did my suggestion to delete the files in /etc and then wanted to copy the files from /root/etc_default back to /etc. THIS WAS A GREAT MISTAKE! Copying back didn't work and now the camera do not start any more normal because of missing these files. But the camera starts and after some time it switches the IR cut filter and turns IR leds on, then always reboot again (watchdog) and do the same continuously. But the temperature (lens and body) is now exactly the same as running a original cloud camera. Maybe there is a hope, that the hacks didn't damaged hardware. 
I have 5 cameras (4 more are on the way from bangg...): At the moment 4 original and one broken. 3 of the original cameras shows a MAC 28:6c:07, which is a address from XIAOMI Elements and one original camera has MAC 5c:ff:ff, which is Shenzhen Kezhonglong Optoelectronic Technology Co., Ltd.
So there are different Wifi modules and/or different hardware used.

Does anybody know without great efforts, to solve my problem with my silly experiment, deleting files from /etc?
Is there any chance to copy files from /root/etc_default to /etc again. Maybe with a script in snx_autorun.sh ? 
Thank you all assistance.

Tommy032017, 2017-03-25:
@samtap Thanks for testing. Pretty strange. I don't know, why there are so many problems by overheating.
When did you get your cameras, last year?
Did you also installed the latest firmware update from MiHome?
Are your pictures clear, or do you get any grainy or blurry effects in night view?

MarkG55, 2017-03-26:
@samtap Are you able to put a USB current meter on both those cameras to see what current they're each drawing?  Roughly, what was the ambient temperature at the time that you did your test?

MarkG55, 2017-03-26:
@Tommy032017 The only suggestion that I could make to recover that camera might be to disassemble the camera and try attaching to the serial console port.

If the camera isn't fully starting up (i.e. it's likely that not all hardware is being switched on, such as the sensor), it's not surprising that it's not heating up as much, so I wouldn't count that as a guarantee that there isn't hardware damage.

samtap, 2017-03-27:
It's a good idea to use the serial console as it allows more debugging, but it's not required. Going by the description of @Tommy032017 a couple posts back, it indeed sounds like the watchdog module isn't unloaded and causes the reboot. So rmmod -f snx_wdt in snx_autorun.sh to unload it and cp -R /root/etc_default/* /etc/ to recover the files. Probably also a good idea to run 'telnetd &' so you can login (don't know if the dropbear/telnet scripts from sd still run). Then it comes down to timing, inserting the sdcard when the boot process is far enough along to auto-mount and run snx_autorun.sh, but before the watchdog reboot kicks in.

Tommy032017, 2017-03-27:
@MarkG55 Thank you. I connected the camera with the serial port, but did only get some hieroglyphics. I tried this with minicom and putty. May be a reason of missing files of /etc.
@samtap Thank you. I also tried to switch off watchdog and then to copy the missing files, but I think, without existing files of /etc, the camera won't be able to mount and/or read the snx_autorun.sh.
I will stop to recover the camera and will keep it as spare parts.

At the moment I'm not really sure to install the hack on a new one, because of the heating problems and coming up blurry and grainy pictures. Hopefully somebody will find out the reason to this. 
I am sure, that you'll find a solution.
Maybe the solution could be found, if there would be a step by step loading (installing) of all the single mods, which are be done automatically at once at the moment. Means to install first telnetd - pause - install next one, and so on. In between the pauses and next loading the behavior (temperature) of the camera could be checked.
Thanks again.

MarkG55, 2017-03-27:
@Tommy032017 I think that your method is the way to go - as long as you've got a current meter on hand to measure the current at each step - makes it much easier to immediately see which step has caused an increase in current consumption (and therefore an increase in heat production)

It might be that you haven't got the right serial port settings - make sure that the baud rate, etc is set correctly!

Tommy032017, 2017-03-27:
@MarkG55 Good idea! I'll agree to your suggestions. I'll order an USB current meter and do some more tests, but first I would like to have a solution to do the hack loading step by step. 

> @Tommy032017 Thanks for that - I've just gone through the same process to reset everything back to how it originally was and it hasn't made any difference for me - the camera is still drawing the [...]
Interesting would be, what still exists after going back. I think, there are changes by the hack, which are still exists further on. Do you have any idea, which processes won't be deleted, or do you think, after going back, everything will be as original?
Do you know, how to realize a script, which will do a step by step installation?

[49 more replies]