samtap/fang-hacks #65: What about Privacy and Spy by using the hack

Tommy032017, 2017-03-18:
Thanks especially to samtap and for everyone involved to this great project.
I would like to use the cameras within Zoneminder or Synology Surveillance Station. Within Synology I don't get any audio at this moment. (Anybody solved the audio problem and could give me a solution to this?)

I think you did the hack to get your own control over the camera without any doubts about privacy and spy by other companies.

I tried to get some more information about the connections to the cloud by using Wireshark. 
I got a lot of different connections to IP addresses with connections to:
- ismartalarm
- xiaomi
- alibaba
- andon.com
- jiuan.com
- aws.amazon.com
- hscctvcn.com  (may be the manufacturer of the SC2135 PCB ?)
and "Beijing Kingsoft Cloud Internet Technology Co., Ltd", which seems to be the P2P, Web-RTC cloud hosting by XIAOMI.

Did you tested the communication to foreign IP addresses, when activating your hack and setting off the cloud communication? Are there all communications, listed above, closed?

I am using since years smartphones from Xiaomi and using the Mi-Cloud. So may be Xiaomi even knows everything about me and my concerning about privacy is still totally unnecessary.

Thanks.

conradhagemans, 2017-03-21:
If you could help me how to test it's not a problem for me to test.
I installed Wireshark already on my laptop.

The local IP Address of my camera is 192.168.1.205

Tommy032017, 2017-03-21:
I don't know why my comment is "ghost"? It was me writing this.
I am using a router (fritzbox), which is capable to capture the whole local traffic with the command:
http://fritz.box/html/capture.html
The exported file *.eth could be opened as a standard file in Wireshark.
Then you need to use this filter ip.addr == 192.168.1.205, which is the IP of the camera. 
Now you'll only see the traffic of the camera and all destinations IP addresses.
I don't know, if you have the possibility to capture your local traffic.
Thanks again.
By the way: Could you please give me some information about your using hack. You wrote, that your camera won't run very hot. My camera increase up to 55°C in night view. Sometimes the stream is purple, if the camera switches back to color mode. 
At the moment I am helpless with the current settings. Pls. see: [https://github.com/samtap/fang-hacks/issues/61](url)
Maybe you could give me a hint to solve the temperature problems.

Tommy032017, 2017-03-30:
Concerning privacy and maybe spy, I have discovered a suspicious initial situation.
One of my cameras was delivered with a MAC 5c:ff:ff..., which belongs to Shenzhen Kezhonglong Optoelectronic Technology Co., Ltd. But downsides on the label of the camera is the MAC 28:6c:07...., which is XIAOMI.
Also this camera is the only one, which will be detected by TinyCam with an P2P stream, but stream will not be shown by the app.
My only conclusion is, that this camera was delivered with a hack and somebody did something with this camera, before it was delivered to me, as a new one.

Cosby86, 2017-05-02:
Hello all, i'm concerned about the privacy too. any update on this topic?

AndiBcool, 2017-05-25:
I did apply the hack, changed to wireless client and disabled cloud applications. 
The cam still does connect to the internet. I got a Yi small Ant Cam with similar hacks running (see https://github.com/fritz-smh/yi-hack for more informations), that does not connect to the web anymore.

My workaround since i use a Fritz Box router too is, activate child restriction and block internet access.
I connect to my ip cams via VPN anyway, so i dont need internet access for the xiaofang.
Anyway this could be different if you wanna stream to the internet.

AndiBcool, 2017-05-26:
Oh ok i just found that the child restriction blocks the streaming, so yeah some blocking of the spy connections would be great..

rushashell, 2017-05-29:
ghost, I solved the audio problem by adding the following 2 lines to the .ExpCam export file: 
`enable_audio = 1
audio_format = 1`.