samtap/fang-hacks #69: gstreamer

mexusbg, 2017-03-22:
I would like to try gstreamer-1.0 to stream from the camera.
It has a very nice client for android. I have problems with RTSP and VLC on Android.

Can you please include it or point me to the right direction how to cross compile it for this camera?
It would be great if I get gstreamer running on this device.

Great project, thanks.

conradhagemans, 2017-03-23:
Cross compile it a bit difficult. We do not have access to the right SDK.
I found an SDK online but cannot get working applications without due to lack of all the drivers.

samtap, 2017-03-24:
It can be compiled but it's useless as there's no hardware accelerated video and the cpu of our device is too slow. Same as ffmpeg.

iothing, 2017-03-28:
I quick and most certainly not complete google search revealed that gstreamer can be configured to just copy the hardware encoded h264 stream w/a re-encoding however most posts implied that this does not really work ... maybe it can be done with some patching.

samtap, 2017-04-29:
Yes, like I said the same as ffmpeg. There's an example included that copies the h264 stream 1:1 and splits it into HLS segments. Even a simple task like that completely exhausts the cpu.

I'm sure gstreamer can do some stuff that ffmpeg can't and vice-versa but they're both the same in general and rely on external codecs, libraries to do stuff like transcoding, on cpu. It's all general-purpose software meant to run on general-purpose hardware. It's completely opposite to what we need.

tinyCam works great for me on Android. If it doesn't work for you it's due to your wifi signal or misconfiguration etc, not because the RTSP server application itself is incompatible.

closed 2017-04-29 by samtap