samtap/fang-hacks #71: Idea: Why not use the P2P stream of the original firmware of the camera?

Tommy032017, 2017-03-22:
Because of my problems overheating the camera, getting bad pictures and SONIX chip will heat up to 70°C, I did a brainstorming and some tests.
All be done by original firmware (think the latest V3) and MiHome APP (4.1.7, Android, Plugin version 2.6.4).

First I installed the camera over the app as an access point by adding a new camera and selecting in the app menu iSmartAlarm Spot HD Video....
I also had to add a new password for this wifi connection (p.e. 12345678).
Then I could see at my smartphone and also by my linux PC the new access point of the camera: isa-camera-isc5_miap9C2A and could connect with the MiHome app and simultaneous with the PC.
Within linux and Wireshark I got these information: 10.42.0.1 = camera access point
```
/tmp/wireshark_pcapng_wlp2s0_20170322222343_27SAee	380	Pakete	insgesamt,	8	angezeigt
				180	15.706765842			10.42.0.100											10.42.0.255											UDP						110				60408	→	32761		Len=68
Frame	180:	110	bytes	on	wire	(880	bits),	110	bytes	captured	(880	bits)	on	interface	0
				Interface	id:	0	(wlp2s0)
				Encapsulation	type:	Ethernet	(1)
				Arrival	Time:	Mar	22,	2017	22:23:59.602180229	CET
				[Time	shift	for	this	packet:	0.000000000	seconds]
				Epoch	Time:	1490217839.602180229	seconds
				[Time	delta	from	previous	captured	frame:	0.381623017	seconds]
				[Time	delta	from	previous	displayed	frame:	0.000000000	seconds]
				[Time	since	reference	or	first	frame:	15.706765842	seconds]
				Frame	Number:	180
				Frame	Length:	110	bytes	(880	bits)
				Capture	Length:	110	bytes	(880	bits)
				[Frame	is	marked:	False]
				[Frame	is	ignored:	False]
				[Protocols	in	frame:	eth:ethertype:ip:udp:data]
				[Coloring	Rule	Name:	UDP]
				[Coloring	Rule	String:	udp]
Ethernet	II,	Src:	b0:e2:35......	(b0:e2:35:......),	Dst:	Broadcast	(ff:ff:ff:ff:ff:ff)
				Destination:	Broadcast	(ff:ff:ff:ff:ff:ff)
								Address:	Broadcast	(ff:ff:ff:ff:ff:ff)
								....	..1.	....	....	....	....	=	LG	bit:	Locally	administered	address	(this	is	NOT	the	factory	default)
								....	...1	....	....	....	....	=	IG	bit:	Group	address	(multicast/broadcast)
				Source:	b0:e2:35:	(b0:e2:35:
								Address:	b0:e2:35	(b0:e2:35:)
								....	..0.	....	....	....	....	=	LG	bit:	Globally	unique	address	(factory	default)
								....	...0	....	....	....	....	=	IG	bit:	Individual	address	(unicast)
				Type:	IPv4	(0x0800)
Internet	Protocol	Version	4,	Src:	10.42.0.100,	Dst:	10.42.0.255
				0100	....	=	Version:	4
				....	0101	=	Header	Length:	20	bytes
				Differentiated	Services	Field:	0x00	(DSCP:	CS0,	ECN:	Not-ECT)
								0000	00..	=	Differentiated	Services	Codepoint:	Default	(0)
								....	..00	=	Explicit	Congestion	Notification:	Not	ECN-Capable	Transport	(0)
				Total	Length:	96
				Identification:	0x5002	(20482)
				Flags:	0x02	(Don't	Fragment)
								0...	....	=	Reserved	bit:	Not	set
								.1..	....	=	Don't	fragment:	Set
								..0.	....	=	More	fragments:	Not	set
				Fragment	offset:	0
				Time	to	live:	64
				Protocol:	UDP	(17)
				Header	checksum:	0xd4d4	[validation	disabled]
								[Good:	False]
								[Bad:	False]
				Source:	10.42.0.100
				Destination:	10.42.0.255
				[Source	GeoIP:	Unknown]
				[Destination	GeoIP:	Unknown]
User	Datagram	Protocol,	Src	Port:	60408	(60408),	Dst	Port:	32761	(32761)
				Source	Port:	60408
				Destination	Port:	32761
				Length:	76
				Checksum:	0xa7db	[validation	disabled]
								[Good	Checksum:	False]
								[Bad	Checksum:	False]
				[Stream	index:	16]

```
Then I installed tinyCam app and let tinyCam search for new cameras: WOW! tinyCam found my XIAOFANG, specified as P2P-UID with 20 characters, and I could save it as a new camera.
Unfortunately I do not get any video stream. I think this will be a problem of some certificates and/or user and password.

I am not a programmer and could not evaluate, if this could be helpful to rethink a new attempt for a different hack solution.
But I think, if only the wifi mode would be changed from access point to client mode the P2P-UID stream will be available at the complete local network.
Next will be to use the existing stream of the original firmware in other applications.
But if tinyCam could use Chinese P2P streams, I see a potential, that the original stream will be usable.
May be tinyCam could handle this stream and also could be used as a NVR. Solution could be an older smartphone, running tinyCam local and using also this app external for life view and playback.

..... only some brainstorming, because of the high temperatures of my cameras. Anyway, thanks to all programmers to their great work to get rid of the cloud service and MiHome.

samtap, 2017-03-24:
The problem is lack of information about the original software, P2P-UID protocol etc. I can't imagine we can make it work but even if we could do something useful with it, it would require reverse engineering everthing without access to the source-code etc, not saying it can't be done but I'm not going to spend time on this approach myself.

D-Pyro, 2017-04-09:
Android app is just java which can be reverse engineered easily if you know what you're doing.

samtap, 2017-04-09:
My bank also has an android app so I'd better reverse engineer that and make my account balance a little more joyful ;-)

Like I said it's not impossible but I'm taking a different approach. Feel free to fork my repo (that's what it's for) and let me know where I can be of assistance.

closed 2017-04-29 by samtap