samtap/fang-hacks #72: rtsp server NOK

eterpstra, 2017-03-23:
Hi,

I am trying to install the fang hack on my camera. After perparing the SD card i am able to login into the status page.

However, RTSP server gives the status NOK. When enabling it it still wont start.

log:

[code]
Tue Oct 11 00:00:31 GMT 2016 - /etc/fang_hacks.sh: Failed to find hacks in /media/mmcblk0p2/data!
mount: mounting /dev/mmcblk0p2 on /media/mmcblk0p2 failed: No such file or directory
Tue Oct 11 00:00:31 GMT 2016 - /etc/fang_hacks.sh: Failed to find /media/mmcblk0p2/data!
Thu Mar 23 16:09:07 GMT 2017 - /media/mmcblk0p1/snx_autorun.sh: Running (device: mmcblk0p1)
Thu Mar 23 16:09:07 GMT 2017 - /media/mmcblk0p1/snx_autorun.sh: Created /tmp/www/cgi-bin
[/code]

It seems that mmcblk0p2 cannot be mounted.

Is there a way to fix this? I formatted the SD card and reset the camera to factory settings but it didnt solved

samtap, 2017-03-24:
Did you try a different sdcard?

nehaleem, 2017-03-26:
I had same issue. I tried to restart the camera (just unplug the cable) with sd card inserted and everything was ok (mounted :-) )

samtap, 2017-03-29:
@eterpstra please join discord chat for support

closed 2017-03-29 by samtap