samtap/fang-hacks #75: Stream runs for short period then seems to hit a bottle neck

kaotix, 2017-03-26:
So I've been testing my stream through motioneyeOS and a few other android APPs directly to the camera and I've found that the RTSP server seems to hit a bottle neck after around 10 seconds of streaming which causes the stream to stop and become unavailable for around 10 seconds then it will come back alive.

The RTSP process itself can be seen using CPU during the actual streaming (around 20~30%) then when it hits the bottle neck it's down to the idle range of around 3~5%
Has anyone else had these symptoms? Not sure if there's something not quite right somewhere?
I only have RTSP/Dropbear/IR Control running, nothing else.
Using a samsung EVO microSD card in the camera (known good)

Thanks

Toomix, 2017-03-27:
Hello,
I use VLC to connect and RSTP stream works fine. I use 64GB Samsung Evo+ card

kaotix, 2017-03-27:
I've tried on VLC as well, get the same result. I have the 32GB version of the same card I think, definitely not the card as I use it in other cameras (one's doing 4k records so know speed/health is fine)

iothing, 2017-03-28:
What does the type of SD card has to do with the rtsp stream? Frames are not buffered on the SD card.
Toomix: I was not able to use the android play store VLC version to access the rtsp stream, how did you do it?

Toomix, 2017-03-29:
I use desktop version of VLC to access the rtsp stream ;)

samtap, 2017-03-29:
It's probably a caching issue on the client side. It's mentioned elsewhere that some of the stream meta-data is incorrect / not present, which could confuse the client. Some work better than others, try adjusting some parameters if your client allows it.

samtap, 2017-04-29:
Closing, unable to reproduce and no response.

closed 2017-04-29 by samtap