samtap/fang-hacks #77: RTSP - Is the hack required WHILE using RTSP

TDPFlorida, 2017-03-29:
I was able to get the RTSP working (ONCE) but I can't reproduce it? Do I need to keep the SD card in (with the hack) for this to work?

Is the an option to ONLY record during motion detection?

Thanks

samtap, 2017-03-30:
Yes the sdcard must remain inserted as it contains all the files needed, i.e. the rtsp server executable. No MiHome doesn't work while the rtsp server is running. You can use other services that don't use camera hardware (ftp, ssh etc). Recording and motion detection are not (yet) supported except by MiHome.

closed 2017-03-30 by samtap

TDPFlorida, 2017-03-30:
Thank you!!!! Is there a manual of the camera online/download so I don't have to bother you with questions about the camera?
Thanks