samtap/fang-hacks #79: Installing on Xiaomi cube camer

adrianmihalko, 2017-03-30:
Hi,

I followed instructions:

1, downloaded .img release and I installed to sd card with Etcher.
2, Rebooted camera, waited for still blue light
3, I inserted the prepared sd card.
4, I checked the status page, it seem to be OK.

Now, when I clicked on Install:

Applying fang hacks scripts...
Running fang_hacks.sh...
Removed stale logfile
Executing script (enabled: 1)
Waiting for cloud apps...
iCamera is running!
Boa webserver is running!
Not linking /media/mmcblk0p1/bootstrap/www/action: already exists
Not linking /media/mmcblk0p1/bootstrap/www/func.cgi: already exists
Not linking /media/mmcblk0p1/bootstrap/www/network: already exists
Not linking /media/mmcblk0p1/bootstrap/www/scripts: already exists
Not linking /media/mmcblk0p1/bootstrap/www/status: already exists
Failed to find hacks in /media/mmcblk0p2/data!
Failed to find /media/mmcblk0p2/data!

and after that:

Error: Nothing is mounted on /media/mmcblk0p2! Manual Mount 

What I am doing wrong?

adrianmihalko, 2017-03-30:
Contents of /tmp/hacks.log:
Thu Mar 30 15:57:47 GMT 2017 - /etc/fang_hacks.sh: Removed stale logfile
Thu Mar 30 15:57:47 GMT 2017 - /etc/fang_hacks.sh: Executing script (enabled: 1)
Thu Mar 30 15:57:47 GMT 2017 - /etc/fang_hacks.sh: Waiting for cloud apps...
Thu Mar 30 15:57:47 GMT 2017 - /etc/fang_hacks.sh: iCamera is running!
Thu Mar 30 15:57:47 GMT 2017 - /etc/fang_hacks.sh: Boa webserver is running!
Thu Mar 30 15:57:47 GMT 2017 - /etc/fang_hacks.sh: Not linking /media/mmcblk0p1/bootstrap/www/action: already exists
Thu Mar 30 15:57:47 GMT 2017 - /etc/fang_hacks.sh: Not linking /media/mmcblk0p1/bootstrap/www/func.cgi: already exists
Thu Mar 30 15:57:47 GMT 2017 - /etc/fang_hacks.sh: Not linking /media/mmcblk0p1/bootstrap/www/network: already exists
Thu Mar 30 15:57:47 GMT 2017 - /etc/fang_hacks.sh: Not linking /media/mmcblk0p1/bootstrap/www/scripts: already exists
Thu Mar 30 15:57:47 GMT 2017 - /etc/fang_hacks.sh: Not linking /media/mmcblk0p1/bootstrap/www/status: already exists
Thu Mar 30 15:57:47 GMT 2017 - /etc/fang_hacks.sh: Failed to find hacks in /media/mmcblk0p2/data!
mount: mounting /dev/mmcblk0p2 on /media/mmcblk0p2 failed: No such file or directory
Thu Mar 30 15:57:47 GMT 2017 - /etc/fang_hacks.sh: Failed to find /media/mmcblk0p2/data!

samtap, 2017-03-31:
The data partition (mmcblk0p2) isn't found. You can check the status page and scroll to the bottom for a list of mounted filesystems, you should see a line for mmcblk0p1 and mmcblk0p2. 

Possible causes are defective sdcard, image not written correctly, resize2fs busy expanding the partition. You can check for resize2fs in the process list on the status page (above the list of mounted filesystems)

closed 2017-03-31 by adrianmihalko