samtap/fang-hacks #80: VLC Record, compatible stream

madmaxkz, 2017-03-31:
Thanks a lot for your project!

I want to use the cam just for one thing. That another person can do record from this cam, without any knowledge of programming. 
VLC-Player plays perfect any stream that I can set on your script, but it does not record any of then. 
It works just in MJPEG, but without a sound. 
That is because a VLC does not really record, just capture. 

Can you help me please? I have no idea how I do this.

samtap, 2017-03-31:
VLC can capture: https://wiki.videolan.org/Documentation:Streaming_HowTo/Receive_and_Save_a_Stream/#Save_a_stream_with_VLC

madmaxkz, 2017-03-31:
Thanks for quick reply. 
I know that VLC can capture, but it works not properly. 
I can see the stream, but when I start a record VLC creates a file, that always 0KB has. (and it's not metter how long a record goes). When I use a MJPEG, I become a videofile, but without a sound.
I think that it hang on a stream, but I have not enough knowladges to do that right (another stream type?).
I have tryed another qualities and resolutions, it works not. When I do that with another source, it works good.

madmaxkz, 2017-04-01:
I tryed with iSpy to record and have another thing. It makes a videofile, but it's just a snapshot that has an duration of the video with sound. 
It's so sad, because live stream works great and the quality is very good.
Anyway, thanks a lot, without hacks would it been not usable (MiHome is worst and have no pc version).

samtap, 2017-04-20:
Though it works for me in VLC on PC and tinyCam on android, it's been mentioned the rtsp server doesn't strictly conform to rtsp spec which may cause problems for some clients? Either way not something I can do anything about, maybe you'll have better luck once we have an improved rtsp server.

closed 2017-04-20 by samtap