samtap/fang-hacks #81: Partitions on SD Card

TDPFlorida, 2017-03-31:
I use the image supplied to CREATE the partitions/files on the SD Card. (I'm using a 32G SD) Everything works fine but when I deselect "Hacks enabled" <APPLY> and <REBOOT> in the MiHome APP it shows only .1G available on the SD card. I've tried expanding the first partition to include the rest of the SD card but once I do that RTSP no longer works (all the FangHacks I'm guessing.) Is there a way to increase the only visible partition so its not wasted?
Thanks

samtap, 2017-04-01:
If you're not happy with the default partition layout you can use a partition manager to change it. For Windows, [MiniTool Partition Wizard](https://www.partitionwizard.com/free-partition-manager.html) works well

Dylantje, 2017-04-01:
Sorry for the newbie question...
I did create no portion just flash the image to my sd... 
What are the thing[s] that i cannot use when i do this?

TDPFlorida, 2017-04-01:
I can just tell you what happened with me when I flashed the SD card with the image provided I was able to bring up the FangHacks URL:
Hacks enabled:
•	RSTP works
•	MiHome (iPhone APP does NOT work with APP features like Recording etc.)
•	NO timestamp on RSTP
Hacks NOT enabled:
•	No RTSP
•	MiHome (iPhone APP does work with APP features like Recording etc.)
•	Timestamp appears in MiHome APP

My problem now is I am using a 32G SD card and when I apply the image to the SD card. MiHome shows the SD card as having .1G so MiHome sees no other partition that the first one. When I use "EPM", I can put all the space in the first partition BUT RTSP no longer works.

I will try using the "MiniTool Partition Wizard" and see if that works.

Any assistance provided would be appreciated!!!

TDPFlorida, 2017-04-02:
I used "MiniTool Partition Wizard" and had 2 partitions: FAT32 (expanded to include all of remaining SD card) and Ext2 (150MB.) That seemed to fix my problems because when I enable the FangHacks, the RTSP works and when the FangHacks are disabled, most of the SD shows up in the MiHome APP. Now if I could just view the data stored on the SD card while its in the camera via the local network?
Thanks samtap!!!

closed 2017-04-02 by TDPFlorida