samtap/fang-hacks #82: Picture shot tests

magadan, 2017-03-31:
Hello, its possible to take a picture (not video) with this camera ?

Will be interesting to have this option and automated to take picture shots of multiple cameras at same time !

Someone can publish a demo pictures to see the quality of this camera ? Thank you

magadan, 2017-03-31:
More one info, im searching if is possible to do a 3D Scan rig with this camera, with 40 units.

I really need to know if is possible to create a interface and take pictures of all cameras, and sent to windows ( smb folder ), or ftp upload. 

Example: http://i.imgur.com/oq4vsLu.jpg

dmolner, 2017-04-01:
Hello,

You can use ispy for take a photo.
This is the quality
webcam.molner.com
The two first  images are resized to 800x600

Best Regards,
David

va escriure:

or mute
the thread

samtap, 2017-04-03:
This is already requested in #39, please put any ideas or other info on that issue

closed 2017-04-03 by samtap