samtap/fang-hacks #83: Is there a way to access SD card from local network...

TDPFlorida, 2017-04-01:
Is there a way to access SD card (without pulling it out and inserting in PC) from network while hack is running or not?

My camera is in a place that's hard to get to (high up) so I cant just pop the card out without using a ladder.

The Xiami Smart 720's I can access the card from a URL (on the network) without any hack.

Thanks

ZeroPoints, 2017-04-02:
Did you not read the main page of this hack. It supports FTP and SSH. Or are you trying to access the device without this hack?

TDPFlorida, 2017-04-02:
Yes, about 50 times but did I miss something? I am able to access the FangHacks page via the network and turn on/off the "hack" or change the timestamp via the network (without directly taking the SD card out and accessing it by inserting it into a PC.) I'd like to be able to ALSO access any recorded videos or time-stamps the same way without taking the SD card out of the camera. Is that possible via FTP or SSH?

ZeroPoints, 2017-04-02:
I don't see why it wouldn't be possible. The cam is like a mini pc. All your doing is accessing its file system through either FTP or SSH. And both are enabled by default when you install this hack

TDPFlorida, 2017-04-02:
Thanks. I am an electrical engineer but I don't have much experience with FTP or SSH. I would appreciate any direction on things to try to get this working. This is the last hurdle before installing 2 cameras on my house to high to reach without a tall ladder.
Thanks

TDPFlorida, 2017-04-02:
You've ALREADY pointed me in the right direction!!! I used "WinSCP" which I used to communicate with my iPhone and now I seem to have access to the "recorded" and "time-lapse" files (when the hack is enabled) on the camera without taking the SD card out. Now if I can figure out a way to JUST record ONLY if motion is detected...now I have it set to "continuous recording" in the MiHome APP which records the entire time.
Thanks

closed 2017-04-02 by TDPFlorida