samtap/fang-hacks #85: Partitions used to store "recordings" and "time-lapse"?

TDPFlorida, 2017-04-02:
There are 2 partitions created by the image supplied: FAT32 and Ext2

FAT32 is the only partition that windows "sees" and the camera creates directories for recordings and time-lapse so can I assume the FAT32 needs to be expanded to contain all the remaining space on the SD card and I should leave "Ext2" alone which is about 150MB?

Thanks

samtap, 2017-04-03:
It's up to you how you want to use the partitions as I told you already on #81. People use it in different ways, but yes in your case (using the cloud app for recordings, access sdcard on a windows pc) you should expand the fat32 partition.

studioBeaver, 2017-04-03:
I'm not shure you need to expand FAT32 partition. In the first screen of the http://device-ip/cgi-bin/status
you will see a button that extends the partition Ext2, since this is the partition used by the hacks.
Those two folders recordings and time-lapse are default folders if you aren't using hacks.
In the end it's up to you, what you want to do.

TDPFlorida, 2017-04-03:
I did some testing on my own where:
1. I took a snapshot of the partitions and used space.
2. Turned the camera on "continuous recording" and did a "time-lapse"
3. Did another snapshot of the partitions and their used space change and noticed ALL change was on the FAT32 partition.

I've noticed when I disable the hacks and make recordings "continuous recording" and "time-lapse" it stores them on the FAT32 partition and,
the motion detection recordings are stored on the cloud?

Does that sound correct?

Is there a way that I can set the camera ONLY to record when motion is detected and those to be saved on the FAT32 partition?

Thanks for all of your help and these "hacks" are working AWESOME!!!

samtap, 2017-04-03:
All your questions concern the cloud app and mi home. I can't help you since I don't use either. You'll have better luck posting on the Xiaomi [forum](http://en.miui.com/forum-52-1.html)

TDPFlorida, 2017-04-03:
Thanks for the direction...I'll see what I can find out from that forum. The whole cloud issue brings up a concern I have about privacy. What are your thoughts about cloud/privacy using these cameras and is there a way through the hacks (or elsewhere) to disable the camera using cloud and NOT "bricking" it or removing the Wi-Fi access and being able to view it via the internet?

Thanks

samtap, 2017-04-03:
Lack of privacy is the reason I don't use them and why this project was created.
The current version only provides the RTSP stream, none of the other features of mi home are implemented yet. If you have more questions you're welcome to join discord chat, this area is meant to be used for bug reports.

closed 2017-04-12 by samtap