samtap/fang-hacks #86: Static ip configuration not supported

samtap, 2017-04-03:
Wifi Client mode is only available with DHCP. Multiple users have requested support for setting a static IP address, i.e. in #60, #61 and on discord.

The cloud app only supports dhcp and originally my scripts were developed to simply mimic cloudapp behaviour. Therefore no support for static ip configuration is available. 

Users can setup a static dhcp lease in their router as interim solution.

A proper solution will be implemented in the next version
- Extend webif to support static ip, netmask, gateway and dns fields.
- Figure out a way to apply it, probably running ifconfig in place of udhcpc

Tommy032017, 2017-04-04:
@samtap Thanks for your hints!
I got it working and I could set a static IP instead DHCP by selecting "Client mode" in network.
Switching back to cloud mode and DHCP is also working.

What I did:
Change the the 01_network in /data/etc/scripts to:

```
#!/bin/sh

source /etc/fang_hacks.cfg

NETWORK_MODE=${NETWORK_MODE:-0}
DISABLE_CLOUD=${DISABLE_CLOUD:-0}

kill_network()
{
  if [ -e "/var/run/wpa_supplicant.pid" -o -d "/var/run/wpa_supplicant" ]; then
    echo "Terminating wpa_supplicant"
    killall wpa_supplicant
  fi

  if [ -e "/var/run/hostapd.pid" -o -d "/var/run/hostapd" ]; then
    echo "Terminating hostapd"
    killall hostapd
  fi
  if [ -e "/var/run/udhcpc.pid" ]; then
    echo "Terminating udhcpc"
    killall udhcpc
  fi
  if [ -e "/var/run/udhcpd.pid" ]; then
    echo "Terminating udhcpd"
    killall udhcpd
  fi
}

start()
{
  echo "Starting Network..."
  rc=0
  case $NETWORK_MODE in
    0)
      echo "Cloud mode: network unmanaged"
      if [ "$DISABLE_CLOUD" -eq 1 ]; then
        echo "WARNING: DISABLE_CLOUD is enabled!"
        echo "WARNING: Network will NOT be operational!"
        rc=1
      fi
      
      if [ -n "$try_recover" ]; then
        ssid="$(cat /etc/config/.wifissid)"
        key="$(cat /etc/config/.wifipasswd)"
        echo "Trying to recover from failed connection (ssid: $ssid)..."
        do_connect 0 "$ssid" "$key"
        rc=$?
      fi
      ;;
    1)
      echo "WiFi Client mode: using wpa_supplicant.conf"
      CFG_WPA="${CFG_WPA:-/media/mmcblk0p2/data/etc/wpa_supplicant.conf}"
      if [ ! -e "$CFG_WPA" ]; then echo "File not found: $CFG_WPA"; return 1; fi
      kill_network

      wpa_supplicant -B -i wlan0 \
        -c /media/mmcblk0p2/data/etc/wpa_supplicant.conf \
        -P /var/run/wpa_supplicant.pid 2>&1
      rc=$?
      if [ $rc -eq 0 ]; then
         
		ifconfig wlan0 192.168.150.97 netmask 255.255.255.0 broadcast 192.168.150.1 2>&1
[... 140 lines cut ...]
fi
```
You have to edit these entries in the 01_network and change the addresses to your own network:

```
ifconfig wlan0 192.168.150.97 netmask 255.255.255.0 broadcast 192.168.150.1 2>&1
	     echo "nameserver 192.168.150.1" > /etc/resolv.conf 2>&1
	     				
		 #udhcpc -i wlan0 -p /var/run/udhcpc.pid -b 2>&1
```
and
```
 ifconfig wlan0 192.168.150.97 netmask 255.255.255.0 broadcast 192.168.150.1 2>&1
	  echo "nameserver 192.168.150.1" > /etc/resolv.conf 2>&1
    
      #udhcpc -i wlan0 -p /var/run/udhcpc.pid -b 2>&1
```
If you now click on "Wireless Client" the IP address will immediately changed to the new address (here 192.168.150.97). To go on and to "Apply" this setting you have first renew the webpage to the new IP address: http://new_IP_address/cgi-bin/network (or ... status). 
Select again the "Access Point" and enter the "Passphrase" and now you can "Apply" your settings permanently with the static IP address.
To go back to DHCP, just select the wireless mode "Cloud" and "Apply".

[image]

**Would be great, if you will make a user-friendly fix for the next version.
Maybe somebody could add a user-friendly setting by the web gui.**

josej69git, 2017-08-21:
hi. how can i edit this file for static ip? access with ftp and telnet and not find 01_network in any directory. thanks for all.

samtap, 2017-08-21:
It's 01-network in /media/mmcblk0p2/data/etc/scripts Please have a Linux machine/vm with sd reader so you can revert changes offline (ext2 filesystem, you can't read it on Windows), in case it doesn't work and locks you out!

Tommy032017, 2017-08-21:
@josej69git
Works exactly like samtap described. Use an explorer with root privilege, p.e. "sudo nemo ". Edit the 01-network as described above, put back sd card to camera (camera always unplugged from power while changing the sd card) and don't forget to apply the "Wireless Client" mode twice.

josej69git, 2017-08-23:
thanks, actually using windows, will probe it in al linux pc