samtap/fang-hacks #88: client mode

gitingear, 2017-04-04:
This project is really helpful; thanks to all that have contributed. I have successfully run cloud and access point modes but client mode is not working for me. Following is the output of 01-network:

```
Terminating hostapd
Connecting to 'Random Network'...
Failed to connect to wpa_supplicant - wpa_ctrl_open: No such file or directory
Failed to connect to wpa_supplicant - wpa_ctrl_open: No such file or directory
wpa_cli ssid failed:
Connection failed: trying to recover...
Starting Network...
Access point mode: using hostapd.conf
Terminating wpa_supplicant
Terminating udhcpd
Hotspot '192.168.0.1' online

```
Any suggestions would be appreciated.

gitingear, 2017-04-04:
comment entered on discord, closing this
(no solution yet)

closed 2017-04-04 by gitingear

gitingear, 2017-04-05:
follow-up: looks like 01-network script not designed to work with network names that have spaces, also I seem to have a bad access point that under some conditions appears to disconnect shortly after initial connection was made

gitingear, 2017-04-16:
was able to connect to ssid with spaces by using latest check-in of www/network, and modifying 01-network as follows:

add after line 135 (EOF):
```
echo "network={" >> /tmp/wpa_tmp.conf
printf "ssid=\"" >> /tmp/wpa_tmp.conf
echo "$2"\" | tr '+' ' ' >> /tmp/wpa_tmp.conf
echo "psk=\"$3\"" >> /tmp/wpa_tmp.conf
echo "}" >> /tmp/wpa_tmp.conf 

```
(my scripting experience is minimal so above code can probably be written in a better way)

change call to wpa_supplicant (line 154 after addition), as follows: 

`  wpa_supplicant -B -i wlan0 -c /tmp/wpa_tmp.conf -P /var/run/wpa_supplicant.pid >/dev/null
`
some additional notes:
1. even with latest check-in, www/network does not seem to sort, this is a minor inconvenience but can be solved by changing the delimiter in iwlist.awk to command, modify www/network to sort and cut based on comma
2. 01-network can be enhanced to also print bssid to wpa_tmp.conf, this also requires change to www/network to pass additional parameter

mazzanet, 2017-05-27:
Looks to be a bug of some sort with wpa_cli - see [here](https://www.raspberrypi.org/forums/viewtopic.php?f=36&t=95084). I've experimented with various combinations of single and double quotes and escaping but can't get wpa_cli to work within a shell script. Writing to wpa_tmp.conf will have to do it seems.