samtap/fang-hacks #90: Adding Azure IoT Hub Support

srgazevedo, 2017-04-06:
Hi all,
From the Original Firmware in the camera, it can detect movement and we can speak to the camera. Do we have some scripts for movement detection + Microphone enablement?

My idea is simple:
1) Being able to detect movement (using an OpenSource Library maybe?)
2) Write it on a log
3) Use Azure-iot-SDK-C to send a message to Azure IoT Hub whenever a movement is detected
4) Record 15seconds video
5) Upload the video to Azure IoT Hub

The Azure IoT Sdk is here: https://github.com/azure/azure-iot-sdk-c

Any help is appreciated :).

samtap, 2017-04-20:
1. Hardware-assisted motion detection is provided (you'll have to monitor some files in /proc)
2. Probably better to skip the log and read /proc stuff directly from your application
3. You'll have to compile the Azure toolkit, I have no idea if it supports ARM and what dependencies it has. If you need some help regarding build environment etc. feel free to join discord.
4. There's currently no way to record, though it shouldn't be difficult. The biggest issue is you can't share the hardware (when recording, rtsp server can't run etc) which is probably not a problem for you.
5. Assuming the Azure toolkit allows this.

It's currently not in my goals to work on this. Feel free to fork the repo and submit PRs when you have something to share!

closed 2017-04-20 by samtap

mhdawson, 2017-04-30:
@srgazevedo have you started workingon this ?  I'm interested in any info you have already figured out about using the speaker/microphone, as well as the hardware motion detection.

srgazevedo, 2017-05-04:
Hello @samtap thanks a lot for your assistance. Azure IoT SDK can be compiled to ARM :). I am trying to figure out how to cross compile it. Do you have any other info about what /proc is and how to monitor it?

@mhdawson I am still working on it but I would definetely prefer to use some open source motion detection in the camera. Maybe something reading the rtsp and detecting motion (all done inside the camera?) skiping the /proc monitor...